RobotLocomotion/drake #13820: Visualize multiple simulations with Drake Visualizer

mntan3, 2020-08-04:
Origin: https://stackoverflow.com/questions/63252779/visualize-multiple-simulations-with-drake-visualizer

It would be useful to be able to visualize multiple simulations simultaneously with Drake Visualizer, but this isn't currently supported

"To the best of my knowledge this isn't currently possible. Rather than targeting a server URL the drake_visualizer is just consuming indiscriminately broadcast messages. There are ways in which we could add more targeted broadcasting/consumption"- @SeanCurtis-TRI

EricCousineau-TRI, 2020-08-06:
\cc @IanTheEngineer per OSRC / ROS2 convo

jwnimmer-tri, 2020-08-06:
I think if you set `LCM_DEFAULT_URL` to a different value for each (visualizer, simulator) pair, the messages will be transmitted on different socket addresses and not interfere with each other.

jwnimmer-tri, 2020-08-11:
The default value for `LCM_DEFAULT_URL` is shown here:
https://github.com/RobotLocomotion/drake/blob/20d799eb1a37cee612de238d1dbccd672196e3b5/lcm/drake_lcm.cc#L22

Using a different port number, for example, would provide for independent process pairs.

```console
env LCM_DEFAULT_URL=udpm://239.255.76.67:7666?ttl=0 bazel-bin/tools/drake_visualizer &
```
or
```console
export LCM_DEFAULT_URL=udpm://239.255.76.67:7666?ttl=0
bazel-bin/tools/drake_visualizer &
bazel-bin/stuff/my_simulator &
```

@SeanCurtis-TRI and @mntan3, it seems to me like `LCM_DEFAULT_URL` is a sufficient tool for the job.  Is it okay if we close this issue now?  (Possibly by transcribing this into the SO post?)

SeanCurtis-TRI, 2020-08-11:
Agreed

mntan3, 2020-08-12:
Yup, this works for me. Thanks!

closed 2020-08-12 by jwnimmer-tri