RobotLocomotion/drake #13828: drake_visualizer: Should fix the weird round-off error artifacts for ground planes

EricCousineau-TRI, 2020-08-06:
When running `drake-visualizer`, it seems like there's some round-off error between parallel + intersecting faces (e.g. a ground plane and the `grid` objects plane).

Example:

```
$ git log -n 1 --oneline --no-decorate
ade13ca064 Add bindings for PlanarJoint (#13813)
$ bazel run //tools:drake_visualizer &
$ bazel run //examples/atlas:atlas_run_dynamics -- --simulation_time=0.
```

Screen cap:
> [image]

FYI @SeanCurtis-TRI @jamiesnape

EricCousineau-TRI, 2020-08-06:
Setting the grid to be wireframe-only fixes the artifact (for ground planes, that is):

> [image]

SeanCurtis-TRI, 2020-08-06:
FTR *technically* changing to wireframe mode doesn't "fix" the artifact, it merely reduces the artifact by orders of magnitude. What you have going on here is [z-fighting](https://en.wikipedia.org/wiki/Z-fighting). As you rotate the camera you can still get flickering on the lines, but since it is a *much* smaller fraction of the rendered pixels it will be less "in-your-face" obnoxious.

That said, I'm fully in favor of simply changing the surface mode by default as actually having a semi-transparent ground plane provides limited value. 

As I see it, there are two paths forward:

1. Change `director` and update `drake`'s build resources to use it. This would require kitware time.
2. Akin to the hack for [fixing the infinite bounding box](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/limit_clipping_range.py), we can add a start up script on the Drake side that toggles it. There are two questions for me on whether this option would work:
    - Would this get reset every time a new scene is loaded?
    - If there is a per-scene reset, is there a callback interface for toggling it again *after* the scene has been loaded?

SeanCurtis-TRI, 2020-08-06:
And *then* I realized that you'd already created a PR.  :-/

EricCousineau-TRI, 2020-08-06:
:grimacing: 

Yeah, went for (2). Nothing special required!

jamiesnape, 2020-08-06:
I am looking to rebuild Director probably next month when I upgrade VTK, so if also wanted to send something to Director, we could pick it up in the not too distant future. Not the first z-fighting issue we have seen.

SeanCurtis-TRI, 2020-08-06:
I like the idea of not having plug-in architecture to do something so simple.

Looking in director it appears that [the `GridItem` defaults to being wireframe](https://github.com/RobotLocomotion/director/blob/master/src/python/director/visualization.py#L1018) but the 
[main window app sets it to "surface with edges"](https://github.com/RobotLocomotion/director/blob/master/src/python/director/mainwindowapp.py#L185). Seems like a relatively *tiny* change in director. We simply remove that one line and we're done.

We've carried this cost for years. If we can resolve it so simply in the next month, I'd say it's worth waiting the one month.

EricCousineau-TRI, 2020-08-06:
I am totally down for fixing this in a more principled fashion in the future.

However, I disagree with waiting another month.
The PR is *small* and should merge today: [image]

Let's just merge it now, then fix it later?

closed 2020-08-06 by EricCousineau-TRI

SeanCurtis-TRI, 2020-08-06:
@jamiesnape Shall I open a new task for us to track the `director`-side change to make sure it's properly on kitware's radar?

EricCousineau-TRI, 2020-08-06:
I'm also cool with re-opening this - I think that pretty much has all the info here?

PR #13829 (2020-08-06): drake_visualizer: Add grid_wireframe for --use_builtin_scripts
Fixes weird round-off issues for wireframes

Resolves #13828

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13829)