RobotLocomotion/drake #13832: Reworking Drake's homepage

jwnimmer-tri, 2020-08-06:
We'd like to rework some of the content and style of https://drake.mit.edu/.  As part of this work, we'll probably need to tweak our workflow for hosting some static assets.  This is a tracking issue to explore that topic.

jwnimmer-tri, 2020-09-16:
Updates:
- The new website style was created using Jekyll.
- The redesign ate the whole site, not just the landing page.

For reference, docs.bazel.build also uses Jekyll via Dockerfile; their code is at https://github.com/bazelbuild/bazel/tree/master/scripts/docs.

Drake's python styleguide also uses Jekyll (since it's based on the Google guide which does so), our preview tool for that is https://github.com/RobotLocomotion/styleguide/blob/gh-pages/preview_site_jekyll.py.

I guess I will see how easy or hard adding Jekyll to our Drake docs pipeline looks like.  An easy local preview is mostly likely the only real challenge.

jwnimmer-tri, 2020-09-18:
My proposal in a quick chat with @jamiesnape today was:

(1) demote the website out of `bazel build //...`;

(2) switch the default sense of `install_prereqs.sh --with-doc` to be opt-in instead of opt-out;

(3) add Ruby and Jekyll as a docs-only dependency on all platforms; on Ubuntu, if we can use the stock apt version of Jekyll, that would be superb;

(4) create a (shadow, non-hyperlinked) page or two in parallel with the live site to stand up the tools and deploy pipeline;

(5) migrate the current content to Jekyll by transmogrifying rst to md using the current prose and contractor's stylesheet;

(6) incorporate the additional page flow and copy editing changes from the contractors.

jamiesnape, 2020-09-18:
`jekyll-sitemap` might be an issue on bionic.

jwnimmer-tri, 2020-09-18:
Yes, I figured we would remove the sitemap entirely until we can change the Documentation builds to Focal.

jwnimmer-tri, 2020-09-22:
@jamiesnape I'm going to put (1..4) above on the To Do board at https://github.com/RobotLocomotion/drake/projects/2 for your queue.

jwnimmer-tri, 2020-10-01:
Related to (1) -- would be good to keep `doc/**` text under pre-merge CI.  So either add a new Publication (without pushing anywhere) build to pre-merge, or add `doc/**` to "everything".  Probably the former.

jwnimmer-tri, 2020-11-02:
Update on (3) -- my new advice is to drop macOS from the platforms we support for local documentation preview.  There are very few contributors who do not have access to Ubuntu.  For Doxygen, Jekyll, and Sphinx, we should only support Ubuntu from now on.

jwnimmer-tri, 2020-11-10:
From #14168 review: it seems like `bazel build //doc:serve_sphinx //bindings/pydrake/doc:serve_sphinx` might be missing from the CI recipe for Documentation builds.  We need to ensure that those program still compile, since the docs instruct developers to run them.

jamiesnape, 2020-11-11:
https://github.com/RobotLocomotion/drake-ci/blob/425c8239e235a46498e682f6facbac43b03b0334/driver/configurations/bazel/step-build.cmake#L58 is the place to look.

jwnimmer-tri, 2020-11-11:
LGTM

jwnimmer-tri, 2020-11-12:
Update on checklist, as I understand it now:

- [x] demote the website out of bazel build //...; update the documentation instructions to make this clear;
- [x] switch the default sense of install_prereqs.sh --with-doc to be opt-in instead of opt-out; update the documentation instructions to make this clear;
- [ ] Remove `--with-doc-only` option from macOS setup scripts.  Document in https://drake.mit.edu/documentation_instructions.html that local preview is only for Ubuntu users.
- [ ] (3) add Ruby and Jekyll as a docs-only dependency on Ubuntu only; we should try very hard to use the stock apt version of Jekyll, even if it means losing the sitemap for now;
- [ ] (4) create a (shadow, non-hyperlinked) page or two in parallel with the live site to stand up the tools and deploy pipeline;
- [ ] (5) migrate the current content to Jekyll by transmogrifying rst to md using the current prose and contractor's stylesheet;
- [ ] (6) incorporate the additional page flow and copy editing changes from the contractors.

BetsyMcPhail, 2020-12-31:
Known issues after testing #14479 (note these are not organized for importance or difficulty).

- Table of Contents missing on: **code_style_tools**, **development_on_mac**, **vim**, **sublime_text**, **reviewable**, **code_style_guide**
- Anchors on all pages missing
- Contents of note and warning boxes aren't formatted correctly (**clion**, **python_bindngs**, Release notes 17 - 20)
-  **clion** green font not implemented
- **clion** too many sections ? Google style guide linting & Drake style addenda should be headers
- **bazel** investigate hyphen no break
- **from_source** link to older_releases.html is broken
- **from_binary** link to release notes (release_notes/older_releases.html#older-releases) is broken
- **vim** download broken
- **vim** "whatever you prefer in your ``.vimrc`` {``init.vim``}" is formatted weirdly
- **release_playbook** doesn't allow nested lists like:
```
1. 
  a.
    i.
    ii.
  b.
```
- **python_bindings** link to Drake’s Tutorials (index.html#tutorials) is broken
- **code_style_guide** [1] footnote not implemented
-  **developers** going to a link and then back doesn't navigate correctly
- **index** binder button icon missing
- release_notes/v0.25.0.html (current) vs release_notes/v0-25-0.html
- release_notes/ vs release_notes.html
- release_notes/older_releases.html vs (new) release_notes/older-releases

jamiesnape, 2021-01-04:
```
Adding
````
permalink: whatever
```
under `title:` may help with some of the edge cases with file names.

Tables of contents may need adding semi-manually if GitHub Pages is missing the plugin. Try https://stackoverflow.com/questions/38417624/table-of-contents-using-jekyll-and-kramdown if so.

jamiesnape, 2021-01-04:
https://kramdown.gettalong.org/syntax.html is also useful.

jwnimmer-tri, 2021-01-26:
As I understand it, now that #14531 + https://github.com/RobotLocomotion/drake-ci/pull/119 have merged, we have a live pipeline where `*.md` files at https://github.com/RobotLocomotion/drake/tree/master/doc/ are published into https://drake.mit.edu/jekyll/ automatically (with the typical ~30 minutes or so delay, i.e., the same as the currently-live website).  So for example:

- https://github.com/RobotLocomotion/drake/blob/master/doc/index.md

is published as

- https://drake.mit.edu/jekyll/

(You can "view source" on the blank page to prove the page is not totally empty.)

jamiesnape, 2021-01-26:
Update is nightly, but otherwise yes.

EricCousineau-TRI, 2021-01-26:
Updated above bullet-points to add in point (5.2), with more info in #14575

jwnimmer-tri, 2021-02-09:
As part of the website transition, for a period of time (once #14577 lands tomorrow) we’re going to have two copies of the website on master (in rst format, and md format).  PRs that change the text of the docs will have to keep both in sync.

EricCousineau-TRI, 2021-02-17:
@ToffeeAlbina-TRI / Jeremy: Who's on point for Python doc Sphinx conversion?
I'm going to call "not it" ;)

jwnimmer-tri, 2021-02-17:
Not sure exactly what you're asking about.  Is it about resurrecting #13983 sans the style-licensing snafu?

My vote would be give up on that and leave the style alone.  To resurrect it, we'd probably want to go back to the contractors to revise the style without including derivative content, but it was such a bear to try to get our patches and macOS unicode font support and what have you fixed up the first time, because the contractors don't use change control, that the bang-for-the-buck there doesn't seem worth it to me.

In any case, I'm not currently planning on working on it.

EricCousineau-TRI, 2021-02-17:
SGTM.

jamiesnape, 2021-02-17:
#13983 would need deconstructing and reconstructing, which is tedious, time consuming, and expensive, and once you have done that you might finally be able to see where the code came from and have to dump it anyway. Same problem as the main site.

jamiesnape, 2021-02-17:
BTW The current theme (RTD) was completely arbitrarily chosen by me, so maybe just find an FOSS one that you like, does not clash too badly, and conveys all the information as best as possible.

jwnimmer-tri, 2021-02-17:
Yes, that's a good point.  If we want to improve pydrake API style, a Drake Developer could just start a new project to find a FOSS style they like better, and get that PR'd.  That's more sensible that trying to rework the prior attempt.  It's a project management question (@ToffeeAlbina-TRI) whether we task anyone to do that, and on what timeframe.

jwnimmer-tri, 2021-02-19:
FYI Here's my command-line to compare the inventory of the two websites, useful for finding pages that are missing or mis-named:
```
jwnimmer@call-cps:~/jwnimmer-tri/drake$ rm -rf ~/tmp/drake-website && bazel run //doc:gen_sphinx -- --out_dir=/home/jwnimmer/tmp/drake-website/sphinx && bazel run //doc:gen_jekyll -- --out_dir=/home/jwnimmer/tmp/drake-website/jekyll && (cd ~/tmp/drake-website/sphinx && find . -name _images -prune -o -name _static -prune -o -type f | sort) > sphinx.txt && (cd ~/tmp/drake-website/jekyll && find . -name images -prune -o -name assets -prune -o -name third_party -prune -o -type f | sort) > jekyll.txt && diff -u sphinx.txt jekyll.txt
```

closed 2021-03-11 by jwnimmer-tri

jwnimmer-tri, 2021-03-11:
The main work is done.  Future improvements are tracked in #14757.

PR #14758 (2021-03-11): doc: Remove the legacy sphinx website
Closes #13832.

Labeled "do not merge" for a few days, so we don't make it more difficult to rollback the website in case of emergency.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14758)