RobotLocomotion/drake #13833: Restructure symbolic files in drake/common

soonho-tri, 2020-08-06:
From https://github.com/RobotLocomotion/drake/pull/13796#issuecomment-667980543

Observation:

 - We have `drake/common/symbolic_*.{h,cc}` files which uses namespace `drake::symbolic` but not in `drake/common/symbolic` directory. This is against GSG.

 - We consolidate all `symbolic_*.{h,cc}` files into a single target, `//common:symbolic` which is used by `//common:default_scalars` and most of Drake code transitively. However,  `//:common:default_scalars` target only needs `symbolic::Expression` and its dependencies.

Plan:

 - Create `drake/common/symbolic` directory and move symbolic files to it.
 - Provide a smaller bazel target which only includes `symbolic::Expression` and its dependencies (including `Variable`, `Formula`, etc).
 - Provide bazel targets for other symbolic files (such as `PolynomialBasisElement`).
 - In the first PR, I will find a way to keep `drake/common/symbolic.h` and `//common:symbolic` target to minimize the changes outside of `common` directory.

/cc @jwnimmer-tri 
/cc @hongkai-dai

soonho-tri, 2020-09-08:
Also check, https://drake.mit.edu/styleguide/cppguide.html#Self_contained_Headers.

jwnimmer-tri, 2022-06-18:
Updated plan: first de-clutter default_scalars by carving out the basic algebra library; later, circle back and split up the extended analysis support libraries.

- Fixes for default scalars:
  - #17419
    - #16394
- Fixes for extended support:
  - TBD

PR #17479 (2022-07-06): [common] Extract remaining symbolic files into smaller libraries
Closes #13833.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17479)
closed 2022-07-06 by jwnimmer-tri