RobotLocomotion/drake #13834: py geometry: Should permit RenderEngine to be extended in Python

EricCousineau-TRI, 2020-08-06:
This permits prototyping towards #11748 

Per convo with @thduynguyen 

FYI @RussTedrake @SeanCurtis-TRI

SeanCurtis-TRI, 2020-08-06:
Could you elaborate on what this entails? Presumably, it's simply a case of [fleshing out the bindings beyond the nominal inclusion that currently exists](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/geometry_py.cc#L93-L99), yes? In essence, resolving the TODO?

If it's more than that, then please elaborate.

EricCousineau-TRI, 2020-08-06:
That, and implementing a trampoline:
https://pybind11.readthedocs.io/en/stable/classes.html#inheritance-and-automatic-downcasting

FWIW Here's the currently generated interface that I'll use to bootstrap my bindings, courtesy of `autopybind11` (FYI @jamiesnape @josephsnyder):
https://github.com/EricCousineau-TRI/repro/commit/aaaefa7501e95986b999395b7487aa22ec61d951

SeanCurtis-TRI, 2020-08-06:
Thanks, the elaboration is *very* helpful.

"Trampoline" is clearly a term of art. Do you have a link to its definition?

jwnimmer-tri, 2020-08-06:
From https://en.wikipedia.org/wiki/Trampoline_(computing), this one seems about right to my experience:

> When interfacing pieces of code with incompatible calling conventions, a trampoline is used to convert the caller's convention into the callee's convention.

EricCousineau-TRI, 2020-08-06:
Also, a little bit nestled away, but:
https://pybind11.readthedocs.io/en/stable/search.html?q=trampoline
https://pybind11.readthedocs.io/en/stable/reference.html?highlight=trampoline#c.PYBIND11_OVERLOAD

PR #13835 (2020-10-21): py geometry: Expose RenderEngine API and trampoline for inheritance
Resolves #13834

FYI @thduynguyen and @SeanCurtis-TRI Review will be coming y'all's way once this is ready ;)

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13835)
closed 2020-10-21 by EricCousineau-TRI