RobotLocomotion/drake #13846: jaco urdf does not visualizer properly in meshcat

RussTedrake, 2020-08-09:
[image]

This is true for both `drake/manipulation/models/jaco_description/urdf/j2s7s300_sphere_collision.urdf` and `drake/manipulation/models/jaco_description/urdf/j2s7s300.urdf`.  Presumably there is some scale getting mis-handled somewhere in the pipeline?

@sammy-tri -- might you take a look?

RussTedrake, 2020-08-09:
I can reproduce it with 
```
./bazel-bin/manipulation/util/geometry_inspector /Users/russt/drake-install/share/drake/manipulation/models/jaco_description/urdf/j2s7s300.urdf --meshcat
```
(this will open meshcat in a different viewer).

Side comments: 
- `show_model` doesn't appear to work well with a meshcat that is opened by the app.  it renders a blank screen.
- It was not clear how to use `show_model` and friends with resources that were fetched from `RobotLocomotion/models`.  When I ran it pointing to the jaco file in the source tree, it complained about missing mesh files.

RussTedrake, 2020-08-09:
interestingly, it seems that the pyplot visualizer falls over, as well.  
```
./bazel-bin/manipulation/util/geometry_inspector /Users/russt/drake-install/share/drake/manipulation/models/jaco_description/urdf/j2s7s300.urdf --pyplot
```

RussTedrake, 2020-09-03:
@gizatt -- fyi, if you are willing to take a look.

gizatt, 2020-09-03:
Problem seems to be the handling of zero-radius spheres (which is the visual geometry given to the base + some other placeholder bodies). Easy fix in both cases -- I'll open a PR.

PR #14013 (2020-09-03): Ignore geometry with invalid sizes in meshcat and pyplot visualizers.
Fixes #13846 by modifying the meshcat and planar scenegraph visualizers to ignore visual geometry with non-positive `geom.float_data` entries.

See #13846 for a picture of `drake/manipulation/models/jaco_description/urdf/j2s7s300.urdf` as visualized in meshcat before this change. After this change, all seems well:
[image]

And the planar visualizer is happy as well (before this change, it threw an error at a convex hull computation during init):
[image]

And the manipulation station looks good too, which I believe stresses a [... cut]
closed 2020-09-03 by RussTedrake