RobotLocomotion/drake #13858: add simulator API element(s) to permit suppression of initialization events

rpoyner-tri, 2020-08-11:
Split from #13741 

Discussions in #13741 proposed adding a parameter to Initialize() to suppress initialization events. This ticket represents the work necessary to design and implement the feature.
- [ ] identify a defensible use case for the feature
- [ ] design the new API element(s)
- [ ] implement the feature for c++, with tests
- [ ] implement the python bindings, with tests

rpoyner-tri, 2020-08-11:
/cc @sherm1 @jwnimmer-tri @EricCousineau-TRI @SeanCurtis-TRI

rpoyner-tri, 2020-08-11:
On reflection, I've change my mind about including this in the milestone. This is a feature request still searching for use case, not a non-repeatability bug.

jwnimmer-tri, 2020-08-11:
We can't presume to know what things a user has defined to happen when the initialize trigger is applied to their Systems.

Because we've coupled two distinct concepts together into the meaning of Initialize (the simulator-internal meaning of resetting of hybrid event timing after updating Context time; and the simulator-external meaning of triggering the System's callbacks defined with that trigger, e.g., clearing a visualizer), and because now we're _forcing_ users to call Initialize to deal with the first meaning, we _must_ provide a way to opt-out of the second meaning, or else the promotion of the warning to an error will be a breaking change with no recourse.

rpoyner-tri, 2020-08-11:
@jwnimmer-tri  From your comment, I gather you claim this is a bug, not a feature request, and therefore should remain in the milestone. Do you have an example of a use case?

jwnimmer-tri, 2020-08-11:
I thought I did that above; maybe I wasn't clear enough.

Imagine the user has defined an Initialize event that resets their visualizer back to some nominal defaults (e.g., some camera view, gridlines, ground plane, etc.).  Imagine the user is setting their context back to time == 0 and showing a few different scenarios against that visualizer.  Imagine the user drags something in the visualizer to see more detail or change the perspective.  Those changes persist even as different scenarios are replayed.  Everything works great.

Now they pull a newer Drake that raises an exception because time changed without an Initialize from one scenario to the next; it says they must call Initialize.  They do so, but now their visualizer keeps resetting itself for every scenario, instead of only the once.

Their only recourse is to remove their initialization event and implement it some other way.  Or, they could opt-out of the event trigger via this issue's new API.

jwnimmer-tri, 2020-08-11:
In other words, evolving from "Initialization events are triggered exactly once, just before time advances" to "Initialization events are triggered whenever time changes outside of the simulator's control" is a breaking change, and we should offer the user a smooth runway between the two different semantics.

sherm1, 2020-08-11:
An alternative to adding a flag might be to separate the two concepts into something like this:
```c++
Initialize() {
  // Trigger initialization events
  StartNewSimulation();
}

StartNewSimulation() {
  // All the initialization stuff except triggering the events.
}
```
Then a user could call either Initialize() or the milder StartNewSimulation() depending on how they want their initialization events dealt with. 

My preference is that a method named "Initialize" ought always to trigger "Initialization Events".

rpoyner-tri, 2020-08-11:
So, y'all, help me understand: how is Nimmer's use case not already addressed by the (admittedly awkward) set_publish_at_initialization()?

jwnimmer-tri, 2020-08-11:
The `set_publish_at_initialization` guards events with a trigger type of `kForced`, and is disabled by default.  We're talking about events with a trigger type of `kInitialization`.

commit 2f988cba (2020-08-18): simulator: Optionally suppress initialization events (#13872)

* simulator: Optionally suppress initialization events

Closes #13858.

 * Add defaulted parameter struct arg to Initialize(). This yields legible call
   sites and leaves room for adding further parameters.
 * Adjust existing test to cover a no-init simulator reuse.
 * Add python bindings, and extend existing tests.
PR #13872 (2020-08-18): simulator: Optionally suppress initialization events
Closes #13858.

 * Add defaulted parameter struct arg to Initialize(). This yields legible call
   sites and leaves room for adding further parameters.
 * Adjust existing test to cover a no-init simulator reuse.
 * Add python bindings, and extend existing tests.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13872)
closed 2020-08-18 by rpoyner-tri