RobotLocomotion/drake #13869: Consider a new macro  DRAKE_MBP_THROW_IF_NOT_PROPERLY_OWNED()

mitiguy, 2020-08-13:
While reviewing PR# 13773 @amcastro-tri made a suggestion about consolidating the various ways to check if an elementType is contained in a MultibodyPlant.  These calls tend to appear in code such as:
 
DRAKE_MBP_THROW_IF_NOT_FINALIZED();
DRAKE_DEMAND(this == &body.GetParentPlant());

  DRAKE_MBT_THROW_IF_NOT_FINALIZED();
  body_B.HasThisParentTreeOrThrow(this);

*******************************************
From Alejandro:
maybe nice if you create a macro similar to DRAKE_MBP_THROW_IF_NOT_FINALIZED with a method in the plant file itself.

Something like:

template <class ElementType>
void MBP::ThrowIfNotOwnedByThisPlant(const ElementType& e, const char* source_method) const {
 if (this != &e.GetParentPlant() )
   throw std::logic_error(
        "Element named '" + e.name() + "' not owned by this plant. From '" + std::string(source_method) + "()'.");
}
and then have the macro:

#define DRAKE_MBP_THROW_IF_NOT_OWNED(e) ThrowIfNotOwnedByThisPlant(e, __func__)
if you follow the pattern for DRAKE_MBP_THROW_IF_NOT_FINALIZED you'll find all the pieces.

jwnimmer-tri, 2021-09-28:
One year later, a grep[1] of the current tree shows exactly one instance of this pattern, introduced during the PR noted atop this issue.  I think a single occurrence is far, far away from a pattern that requires macro sugar.  I suggest closing this issue as being not relevant.

[1] `find . -type f -exec grep --color -nH -e 'DRAKE.*this.*GetParentPlant.*' {} +`

xuchenhan-tri, 2022-05-24:
It seems like that is also the only place where `GetParentPlant` is even used in our C++ code base. I'd say it's safe to rest the case for the MACRO sugar. WDYT @mitiguy ?

closed 2022-05-25 by amcastro-tri