RobotLocomotion/drake #13875: Improve Schunk WSG geometry (e.g. in ManipulationStation)

RussTedrake, 2020-08-15:
The visual geometry of our current public Schunk model is a bit embarrassing:
http://people.csail.mit.edu/russt/uploads/planar_iiwa_teleop.html

It consists of 3 boxes, which appear to float off the end of the robot.  We have considerably better visualizations in anzu, but we need to think about which is most appropriate for the public example (possibly removing the capacitive finger appendage, etc).  

@hjsuh94 -- maybe you can own this?  The folks I've asked recommend that talking to @alexalspach, who generated most of those models, is probably a good first step.

RussTedrake, 2020-08-15:
ftr — worth keeping the mesh file size in mind, too, as the meshcat workflow means folks will download the mesh via websockets every time they open the visualizer.

hjsuh94, 2020-08-16:
This is interesting, but seems like there is no detailed sdf model of the original fingers anywhere. Will talk with @alexalspach and generate some meshes soon.

RussTedrake, 2021-09-19:
This should have been closed with Terry's PR.

closed 2021-09-19 by RussTedrake