RobotLocomotion/drake #13888: Fix MBPlant's CalcReactionForce() and make it maintainable

sherm1, 2020-08-17:
MBP has a joint-reaction force output port that uses a private method `CalcReactionForce()` when in need of computation. Reaction forces are essentially a side effect of our dynamics computations, so `CalcReactionForce()` begins by evaluating forward dynamics, for which there are different implementations depending whether we are in discrete or continuous mode. Currently the reaction force algorithm then tries to reproduce the forces that were used in forward dynamics, feed those to inverse dynamics, and extract the reaction forces from there.

Currently `CalcReactionForce()` fails to capture all the forces that were used in discrete mode due to recent independent changes in the discrete model, making it clear that the current structure is not maintainable. There must be a way for the forward dynamics solvers (discrete or continuous) to control the forces that are used for calculating reaction forces so that they are always correct. This is easy in continuous mode since the articulated body algorithm computes reaction forces as a side effect anyway -- we should be taking advantage of that. But we need to engineer a way for the discrete solvers to define the appropriate reaction forces for their implemented models also so this doesn't break again in the future.

Related to #12769.

amcastro-tri, 2020-08-17:
To be clear, the bug is in **continuous mode**, not in discrete mode. Since in continuous mode we forgot to add the hydroelastic forces computation. 

Master today does do the correct computation in discrete mode.

sherm1, 2020-08-17:
> To be clear, the bug is in continuous mode, not in discrete mode. 

I see `CalcReactionForces()` calls  `CalcAndAddContactForcesByPenaltyMethod()`  which is not called by Tamsi but does that automatically include both point and hydroelastic forces?

amcastro-tri, 2020-08-17:
true, not called by TAMSI, because forces in TAMSI are implicit (so the explicit expression doesn't help). However TAMSI uses the same model and therefore they are consistent. The slight differences when calling on t_previous or t_next is what causes the O(dt) differences Rick observed in the past.
Do we call hydro forces computation in discrete mode? no, we do not support discrete hydro in master yet.

xuchenhan-tri, 2022-05-03:
This issue deserves an update now that we had long since started supporting discrete hydroelastics. It seems like the reaction force is now bugged for both continuous and discrete mode, and the quick (but probably not the best) fix is to add the contribution of hydro in `CalcAndAddContactForcesByPenaltyMethod()`? @sherm1 @amcastro-tri could you comment?

sherm1, 2022-05-03:
This suggests we're missing reaction force unit tests that would squawk when we get this wrong!

sherm1, 2022-05-03:
@amcastro-tri I think we need to get SAP reaction forces right as part of saying we have released SAP in Drake. WDYT?

xuchenhan-tri, 2022-05-25:
Somehow I missed the reply above among the fire hose of github emails.

Looking at this again, it seems like it's reasonable to do the following:

1. Set up a unit test to catch the bug in both continuous and discrete mode.
2. Rename `MbP::CalcSpatialContactForcesContinuous` to simply `CalcSpatialContactForces` since now the computation makes sense for both continuous and discrete.
3. Rename `MbP::CalcAndAddContactForcesByPenaltyMethod` to something like `AddPointContactForce` to better parallel its hydro counterpart.
4. In `MbP::CalcReactionForces` call `CalcSpatialContactForces` instead of `CalcAndAddContactForcesByPenaltyMethod` so that all contact forces are accounted for, continuous and discrete.

amcastro-tri, 2022-05-31:
I think we must first work on #16955. The long term plan is that everything discrete lives in the manager, making things more modular and also cleaner.

I want to move also TAMSI (or its newer version really) into the manager as well. So I don't think it's worth fixing this now unless it is cause of headaches to our users.

xuchenhan-tri, 2022-07-20:
Ok SGTM. I'll de-prioritize this for now then.

DamrongGuoy, 2022-10-05:
@JoseBarreiros-TRI I think your project will have to wait for the fix of this issue, right?

EricCousineau-TRI, 2022-11-17:
Just to check, is  this at all related to @RussTedrake's TAMSI force bug mentioned in Drake Slack on 2022-11-01?
https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1667318747045599

RussTedrake, 2022-11-17:
My issue is slack was with Sap, not TAMSI, and @amcastro-tri was able to reproduce and has been working on resolving it.

EricCousineau-TRI, 2022-11-17:
Gotcha. Sorry, I saw "TAMSI and SAP are different" in the thread, but upon rereading I see "TAMSI seems more correct than SAP" for that post.
Thanks for clarification!

PR #19920 (2023-08-15): Pipes constraints forces through and fixes reaction forces computation
This PR closes #19435 and #19068
It makes use of the infrastructure at the SAP solver level to report constraint forces within SapDriver so that we can compute the aggregate multibody forces due to constraints. With this information, we can reliably compute reaction forces including all terms in the discrete update.

With this PR we include discrete hydro forces along with every other solver constraint in the computation of reaction forces, effectively resolving #19068.

In summary
 - Closes #18749
 - Closes #19068
 - Close [... cut]
commit 4b1dd829 (2023-08-15): Pipes constraints forces through MbP and fixes the computation of reaction forces. (#19920)

It makes use of the infrastructure at the SAP solver level to report constraint forces within SapDriver so that we can compute the aggregate multibody forces due to constraints.
With this information, we can reliably compute reaction forces including all terms (force elements and constraints, including contact) in the discrete update.
This PR also resolves #13888 making the computation of reaction forces maintainable. New constraints will immediately be included in reaction forces.

With this PR we als [... cut]
closed 2023-08-15 by amcastro-tri