RobotLocomotion/drake #13903: Parsing zero mass bodies.

amcastro-tri, 2020-08-19:
This is from a [Slack discussion](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1597783243328300).

When using SDF parsing @avalenzu gets the "The resulting spatial inertia ... is not physically valid" error. However he can create a zero spatial inertia with mass=0, p_PScm_E=[0, 0, 0] and G_SP_E=UnitInertia([0, 0, 0]).

### The problem

`ExtractSpatialInertiaAboutBoExpressedInB()` in `detail_sdf_parser.cc` makes use of `MakeFromCentralInertia()`, which assumes `mass > 0`

**Note I**, zero spatial inertias are supported in our math, that is why programmatically creating zero spatial inertias is allowed.

**Note II**, `MakeFromCentralInertia()` assumes `mass>0`, however this is missing proper docs and throws from somwehere else buried in the code.

### A solution

We can make `ExtractSpatialInertiaAboutBoExpressedInB()` to programmatically create a zero spatial inertia as Andres did when we have `mass = 0`.

### Question

- Do we have the same problem in URDF?
- Does sdformat allow `m=0`? (I (Alejandro) remember at some point in the past there was some hardcoded minimum value).

avalenzu, 2020-08-19:
@mitiguy, as per the discussion @amcastro-tri mentioned, I've assigned you for the doc/throw fix to `MakeFromCentralInertia()`.

avalenzu, 2020-08-19:
@amcastro-tri, in the case of `mass == 0` and non-zero moments/products of inertia, should we take the "zero spatial inertia" branch? Or call `MakeFromCentralInertia()` and let it fail? I ask because the SDFormat default for `<inertia/>` is `ixx = iyy = izz = iyz = ixz = ixy = 1`, so a tag like this:
```.xml
<inertial>
  <mass>0</mass>
</inertial>
```
would give us zero mass and non-zero moments/products. Given that the default for `<mass>` is 1 (i.e. to get `mass == 0` the file has to specify it explicitly), I'd prefer to take the "zero spatial inertia" branch.

amcastro-tri, 2020-08-19:
I actually concur with you, if I say `<mass>0</mass>` only, I'd personally like to take the zero spatial inertia branch.

On further thought, it'd seem wrong, since non-zero inertia components Ixx, etc. would imply non-zero mass. In the end, I believe we should abide by the SDF specification, which now says nothing about this.

Maybe a middle ground would be to make it now so that we do need to say `Ixx=Iyy=etc=0` and work with OSRC to include this case in the spec?

avalenzu, 2020-08-19:
OK, I'll make it so that we take the zero spatial inertia branch iff the mass and all components of inertia are zero.

sherm1, 2020-08-19:
Can we make the case mass=0, inertia≠0 an error? It doesn't make any physical sense.

amcastro-tri, 2020-08-19:
I agree with you @sherm1, but it'd definitely be nice if `<mass>0</mass>` alone meant "zero mass body". But that I believe should be part of the actual SDF spec.

jwnimmer-tri, 2021-11-12:
I'm having a difficult time following this discussion.  Did #13908 resolve the problem?  If not, what is the unresolved victory condition?

sherm1, 2021-11-12:
From above it looks like `SpatialInertia::MakeFromCentralInertia()` still needs to document the requirement that mass > 0 and might need a better error message in that case? What do you think @mitiguy (read above discussion)?

jwnimmer-tri, 2022-05-02:
@mitiguy please share your thoughts.

mitiguy, 2022-05-02:
I agree with Rico.  I have be looking at the URDF/SDF parser code and associated tests and think that the testing for the mass/inertia conditions centralized, so each format can leverage the general code.  

I think the tests should be as exhaustive as possible and the error messages as instructive as possible. 

 In addition, I am working with Open Robotics to update the <inertial> tag and sub-tags as the specification is ambiguous.  Lastly, I have investigated the URDF/SDF from the CAD packages OnShape and am doing the same from SolidWorks.  I have found inconsistencies with our Drake convention and think we need to clearly document our product of inertia conventions.

amcastro-tri, 2022-05-03:
The actual issue Andres had here was resolved in #13908.
Error reporting of zero mass links is better tracked now by #17081.

closed 2022-05-03 by amcastro-tri