RobotLocomotion/drake #13905: benchmark measurment tracking: how to capture compiler in use (and maybe flags)?

rpoyner-tri, 2020-08-19:
I made an attempt to capture compiler-in-use data in the first cassie_bench PR, but failed and had to retreat. This ticket is for consideration of better methods.

Possibilities:
 * preprocessor symbol capture in source/object code
 * compiler-implemented capture to object-code or DWARF tables (e.g. -{fg}record-gcc-switches)
 * reporting via bazel rules (tried this in the PR and failed)

Whatever gets implemented, it needs to at least cover the currently supported platform/compiler combinations, and have a plausible story for extension to others.

rpoyner-tri, 2020-08-19:
https://github.com/rpoyner-tri/drake/commit/09d298330e7b6a7df9a5dd2be2cc058cc09ae7e2 is simple and probably more than fine.

Based on other strategy discussions, could take or leave this issue/patch. /cc @sherm1 @jwnimmer-tri

sherm1, 2020-08-19:
I submitted #13906 to get identify_compiler built -- looks a lot like your change there although without the visibility directive, PTAL.

PR #13909 (2020-08-19): cassie_bench: Record compiler name and version
Closes #13905.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13909)
commit b0fe45a6 (2020-08-19): cassie_bench: Record compiler name and version (#13909)

* cassie_bench: Record compiler name and version

Closes #13905.
closed 2020-08-19 by rpoyner-tri