RobotLocomotion/drake #13918: `drake_jupyter_py_binary` failing immediately on mac

RussTedrake, 2020-08-21:
#13901 reverted my JointSliders commit because it was failing on mac.  I've reproduced.  But it appears(?) to have nothing whatsoever to do with my PR?

Adding the following lines
```
drake_jupyter_py_binary(
    name = "Untitled",
    add_test_rule = 1,
    deps = [
        ":plant_py", 
    ],
)
```
to `bindings/pydrake/multibody/BUILD.bazel` and an empty `Untitled.ipynb`file (created by running jupyter notebook, creating a new file, and saving it immediately) causes the same error.  If I remove the `plant_py`, I get a similar error, but complaining that there is no module named `pydrake`, so I've put it back in.

**I could really use some help on this.**  

```
Last login: Thu Aug 20 19:22:25 on ttys005
russt@Russ-2020-Macbook-Air ~ % cd drake/bindings/pydrake/multibody
russt@Russ-2020-Macbook-Air multibody % bazel test ...
INFO: Invocation ID: 7546aff1-7ca1-4d5d-b6f6-00a39bda4f27
INFO: Analyzed 85 targets (0 packages loaded, 0 targets configured).
INFO: Found 42 targets and 43 test targets...
FAIL: //bindings/pydrake/multibody:py/Untitled_test (see /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test_attempts/attempt_1.log)
FAIL: //bindings/pydrake/multibody:py/Untitled_test (see /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test_attempts/attempt_2.log)
FAIL: //bindings/pydrake/multibody:py/Untitled_test (see /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test.log)

FAILED: //bindings/pydrake/multibody:py/Untitled_test (Summary)
      /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test.log
      /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test_attempts/attempt_1.log
      /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test_attempts/attempt_2.log
INFO: From Testing //bindings/pydrake/multibody:py/Untitled_test:
==================== Test output for //bindings/pydrake/multibody:py/Untitled_test:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 64, in <module>
    import msvcrt
ModuleNotFoundError: No module named 'msvcrt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/sandbox/darwin-sandbox/2101/execroot/drake/bazel-out/darwin-opt/bin/bindings/pydrake/multibody/py/Untitled_test.runfiles/drake/bindings/pydrake/multibody/Untitled_jupyter_py_main.py", line 4, in <module>
    from drake.tools.jupyter.jupyter_bazel import _jupyter_bazel_notebook_main
  File "/private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/sandbox/darwin-sandbox/2101/execroot/drake/bazel-out/darwin-opt/bin/bindings/pydrake/multibody/py/Untitled_test.runfiles/drake/tools/jupyter/jupyter_bazel.py", line 6, in <module>
    from jupyter_core.command import main as _jupyter_main
  File "/usr/local/lib/python3.8/site-packages/jupyter_core/command.py", line 17, in <module>
    from subprocess import Popen
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 71, in <module>
    import selectors
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/selectors.py", line 11, in <module>
    import math
ImportError: ImportError:

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.8 from "/usr/local/opt/python@3.8/bin/python3.8"
  * The NumPy version is: "1.19.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.
[... 234 lines cut ...]
  <frozen importlib._bootstrap_external>(783): exec_module
  <frozen importlib._bootstrap>(671): _load_unlocked
  <frozen importlib._bootstrap>(975): _find_and_load_unlocked
  <frozen importlib._bootstrap>(991): _find_and_load
  /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/sandbox/darwin-sandbox/2105/execroot/drake/bazel-out/darwin-opt/bin/bindings/pydrake/multibody/py/Untitled_test.runfiles/drake/tools/jupyter/jupyter_bazel.py(6): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(783): exec_module
  <frozen importlib._bootstrap>(671): _load_unlocked
  <frozen importlib._bootstrap>(975): _find_and_load_unlocked
  <frozen importlib._bootstrap>(991): _find_and_load
  /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/sandbox/darwin-sandbox/2105/execroot/drake/bazel-out/darwin-opt/bin/bindings/pydrake/multibody/py/Untitled_test.runfiles/drake/bindings/pydrake/multibody/Untitled_jupyter_py_main.py(4): <module>

================================================================================
INFO: Elapsed time: 4.501s, Critical Path: 3.99s
INFO: 6 processes: 2 remote cache hit, 4 darwin-sandbox.
INFO: Build completed, 1 test FAILED, 2 total actions
//bindings/pydrake/multibody:py/Untitled_test                            FAILED in 3 out of 3 in 1.3s
  Stats over 3 runs: max = 1.3s, min = 1.1s, avg = 1.2s, dev = 0.1s
  /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test.log
  /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test_attempts/attempt_1.log
  /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/execroot/drake/bazel-out/darwin-opt/testlogs/bindings/pydrake/multibody/py/Untitled_test/test_attempts/attempt_2.log

INFO: Build completed, 1 test FAILED, 2 total actions
russt@Russ-2020-Macbook-Air multibody %
```

RussTedrake, 2020-08-27:
I have an ugly work-around for this that has me unstuck for the moment.  Downgrading from priority high to medium.

BetsyMcPhail, 2020-09-02:
I can reproduce this on my machine. If I move `Untitled.ipynb` to the examples subdirectory, I don't get an error. Likewise, if I move `door_hinge_inspector.ipynb` up to the `multibody` directory, e.g. 

```
drake_jupyter_py_binary(
    name = "door_hinge_inspector",
    add_test_rule = 1,
    deps = [
        ":plant_py",
        ":tree_py",
    ],
)
```

I see the same mysterious error.

jwnimmer-tri, 2020-09-02:
FWIW here's a comment that I made in the #buildcop slack thread about this:

> It looks to me like `import subprocess` does `import selectors` which does `import math` which is somehow finding `pydrake.math` instead of cypthon's `math`.

BetsyMcPhail, 2020-09-03:
Turns out we've run into this issue before, see #12960:

```
When running under our Jupyter infrastructure, apparently . is on the path, so when we try to import math wanting the Python standard library, we accidentally pick up pydrake.multibody.math instead.

To unbreak the build for how, we'll relocate the notebook to an examples subdirectory.
```

PR #14017 (2020-09-06): Add jupyter widget version of JointSliders
Restores #13884

With one modification: the Jupyter notebook is moved to the  ``examples`` subdirectory. It seems that the Jupyter infrastructure includes the current directory in the path. When the notebook is in the ``pydrake/multibody`` directory, this causes ``import math`` to use ``pydrake.multibody.math`` rather than the Python standard library.

Follow-up to: #13901
Resolves #13918

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14017)
closed 2020-09-06 by EricCousineau-TRI