RobotLocomotion/drake #13924: cart-pole example throws exception in cmake project

Maverobot, 2020-08-23:
Using the same example code (cart-pole) in a cmake project, turned out to throw an exception which did not happen within `drake` repository.

> terminate called after throwing an instance of 'std::logic_error'
>  what():  Can't acquire pose port for unknown source id: 9296016.

To reproduce it, 

* Install drake with binary release

  ```sh 
  wget https://github.com/RobotLocomotion/drake/releases/download/v0.22.0/drake-20200817-bionic.tar.gz 
  sudo tar -xvzf drake-20200817-bionic.tar.gz -C /opt/
  ```

* Download my example cmake project and compile it
  ```sh
  wget https://github.com/RobotLocomotion/drake/files/5115188/drake_examples.tar.gz
  tar -xvf drake_examples.tar.gz
  cd drake_examples && mkdir build 
  cd build && cmake .. && make -j4
  ```

* Reproduce the error
  ```sh
  ./cart_pole/cart_pole_passive_simulation
  ```

sammy-tri, 2020-08-24:
Assigning to @jamiesnape for now, as you're one of our resident experts on how the cmake vs. bazel builds differ.

jamiesnape, 2020-08-24:
May be more a binary release issue than a CMake issue. Were you compiling with SNOPT support when building the drake repository from source? That is one possible difference.

Maverobot, 2020-08-24:
It might be worth mentioning that on the [drake documentation website](https://drake.mit.edu/doxygen_cxx/group__solvers.html), it says

> Note that Drake's pre-compiled binary releases provide SNOPT.

Maverobot, 2020-08-28:
Is there anything which I can do to help fixing this issue?

jwnimmer-tri, 2020-08-28:
It works fine for me:
```
cd $HOME
mkdir tmp && cd tmp
wget https://github.com/RobotLocomotion/drake/releases/download/v0.22.0/drake-20200817-bionic.tar.gz 
tar -xvzf drake-20200817-bionic.tar.gz
wget https://github.com/RobotLocomotion/drake/files/5115188/drake_examples.tar.gz
tar xf drake_examples.tar.gz 
cd drake_examples
mkdir build && cd build
cmake .. -Ddrake_DIR=$HOME/tmp/drake/lib/cmake/drake/
make -j2
./cart_pole/cart_pole_passive_simulation 
```

Maverobot, 2020-08-28:
> It works fine for me:
>
> ```
> cd $HOME
> mkdir tmp && cd tmp
> wget https://github.com/RobotLocomotion/drake/releases/download/v0.22.0/drake-20200817-bionic.tar.gz
> tar -xvzf drake-20200817-bionic.tar.gz [...]

I followed this shell script snippet and got the same error as I posted above.

jwnimmer-tri, 2020-08-28:
Are you using the correct library paths?
```
jwnimmer@cons:~/tmp/drake_examples/build$ ldd cart_pole/cart_pole_passive_simulation 
	linux-vdso.so.1 (0x00007fff01ffa000)
	/home/jwnimmer/tmp/drake/lib/libdrake.so (0x00007fb11a1c7000)
	libgflags.so.2.2 => /usr/lib/x86_64-linux-gnu/libgflags.so.2.2 (0x00007fb119fa2000)
	/home/jwnimmer/tmp/drake/lib/libdrake_marker.so (0x00007fb11c503000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb119c19000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fb119a01000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb119610000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb119272000)
	libvtkRenderingOSPRay-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkRenderingOSPRay-8.2.so.1 (0x00007fb119022000)
	libembree3.so.3 => /home/jwnimmer/tmp/drake/lib/libembree3.so.3 (0x00007fb1187a7000)
	libospray.so.0 => /home/jwnimmer/tmp/drake/lib/libospray.so.0 (0x00007fb118577000)
	libospray_common.so.0 => /home/jwnimmer/tmp/drake/lib/libospray_common.so.0 (0x00007fb118350000)
	libospray_module_ispc.so.0 => /home/jwnimmer/tmp/drake/lib/libospray_module_ispc.so.0 (0x00007fb117cc2000)
	libvtkRenderingVolumeAMR-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkRenderingVolumeAMR-8.2.so.1 (0x00007fb117ab3000)
	libvtkFiltersAMR-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkFiltersAMR-8.2.so.1 (0x00007fb11788c000)
	libvtkParallelCore-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkParallelCore-8.2.so.1 (0x00007fb117636000)
	libvtkRenderingVolumeOpenGL2-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkRenderingVolumeOpenGL2-8.2.so.1 (0x00007fb1173b2000)
	libvtkImagingMath-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkImagingMath-8.2.so.1 (0x00007fb11716f000)
	libvtkRenderingVolume-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkRenderingVolume-8.2.so.1 (0x00007fb116ae6000)
	libvtkImagingCore-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkImagingCore-8.2.so.1 (0x00007fb116697000)
	libvtkIOXML-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkIOXML-8.2.so.1 (0x00007fb11639f000)
	libvtkIOXMLParser-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkIOXMLParser-8.2.so.1 (0x00007fb116186000)
	libvtkRenderingSceneGraph-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkRenderingSceneGraph-8.2.so.1 (0x00007fb115f6f000)
	libvtkIOGeometry-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkIOGeometry-8.2.so.1 (0x00007fb115c5c000)
	libvtkIOLegacy-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkIOLegacy-8.2.so.1 (0x00007fb1159a5000)
	libvtkIOCore-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkIOCore-8.2.so.1 (0x00007fb11573d000)
	libvtkdoubleconversion-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkdoubleconversion-8.2.so.1 (0x00007fb11552d000)
	libvtklzma-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtklzma-8.2.so.1 (0x00007fb1152fc000)
	libvtkRenderingOpenGL2-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkRenderingOpenGL2-8.2.so.1 (0x00007fb114f11000)
	libvtkRenderingCore-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkRenderingCore-8.2.so.1 (0x00007fb114b2b000)
	libvtkCommonColor-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonColor-8.2.so.1 (0x00007fb11490a000)
	libvtkFiltersGeometry-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkFiltersGeometry-8.2.so.1 (0x00007fb114696000)
	libdrake_ignition_math.so => /home/jwnimmer/tmp/drake/lib/libdrake_ignition_math.so (0x00007fb11c4c0000)
	libdrake_ibex.so => /home/jwnimmer/tmp/drake/lib/libdrake_ibex.so (0x00007fb1140fd000)
	libgfortran.so.4 => /usr/lib/x86_64-linux-gnu/libgfortran.so.4 (0x00007fb113d1e000)
	libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007fb113ade000)
	libvtkIOImage-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkIOImage-8.2.so.1 (0x00007fb1137a6000)
	libvtkDICOMParser-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkDICOMParser-8.2.so.1 (0x00007fb11358c000)
	libvtkmetaio-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkmetaio-8.2.so.1 (0x00007fb11330c000)
	libdrake_lcm.so => /home/jwnimmer/tmp/drake/lib/libdrake_lcm.so (0x00007fb11c4a5000)
	libvtkFiltersSources-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkFiltersSources-8.2.so.1 (0x00007fb11307b000)
	libvtkFiltersGeneral-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkFiltersGeneral-8.2.so.1 (0x00007fb112bb0000)
	libvtkCommonComputationalGeometry-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonComputationalGeometry-8.2.so.1 (0x00007fb11297f000)
	libvtkFiltersCore-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkFiltersCore-8.2.so.1 (0x00007fb1123bc000)
	libvtkCommonExecutionModel-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonExecutionModel-8.2.so.1 (0x00007fb112102000)
	libvtkCommonDataModel-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonDataModel-8.2.so.1 (0x00007fb111b51000)
	libvtkCommonMisc-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonMisc-8.2.so.1 (0x00007fb111933000)
	libvtkCommonSystem-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonSystem-8.2.so.1 (0x00007fb11171b000)
	libvtkCommonTransforms-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonTransforms-8.2.so.1 (0x00007fb1114e8000)
	libvtkCommonMath-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonMath-8.2.so.1 (0x00007fb1112c6000)
	libvtkCommonCore-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtkCommonCore-8.2.so.1 (0x00007fb110d37000)
	libvtksys-8.2.so.1 => /home/jwnimmer/tmp/drake/lib/libvtksys-8.2.so.1 (0x00007fb110af8000)
	libdrake_spdlog.so => /home/jwnimmer/tmp/drake/lib/libdrake_spdlog.so (0x00007fb11c422000)
[... cut ...]
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fb109f86000)
	libpord_seq-5.1.2.so => /usr/lib/x86_64-linux-gnu/libpord_seq-5.1.2.so (0x00007fb109d70000)
	libesmumps-6.so => /usr/lib/x86_64-linux-gnu/libesmumps-6.so (0x00007fb109b6b000)
	libscotch-6.so => /usr/lib/x86_64-linux-gnu/libscotch-6.so (0x00007fb1098db000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007fb1096d4000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fb1094bf000)
	libscotcherr-6.so => /usr/lib/x86_64-linux-gnu/libscotcherr-6.so (0x00007fb1092bd000)
```

Maverobot, 2020-08-28:
Instead of under `$HOME` I put the stuffs under `/tmp`

```
❯ ldd cart_pole/cart_pole_passive_simulation 
	linux-vdso.so.1 (0x00007ffe6f9dd000)
	/tmp/tmp/drake/lib/libdrake.so (0x00007f404e913000)
	libgflags.so.2.2 => /usr/lib/x86_64-linux-gnu/libgflags.so.2.2 (0x00007f404e6ee000)
	/tmp/tmp/drake/lib/libdrake_marker.so (0x00007f40509fe000)
	libdrake_ignition_math.so => /tmp/tmp/drake/lib/libdrake_ignition_math.so (0x00007f40509c5000)
	libdrake_lcm.so => /tmp/tmp/drake/lib/libdrake_lcm.so (0x00007f40509ac000)
	libdrake_spdlog.so => /tmp/tmp/drake/lib/libdrake_spdlog.so (0x00007f405092d000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007f404e4da000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f404e2bb000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f404dee7000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f404db49000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f404d931000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f404d540000)
	libvtkRenderingOSPRay-8.2.so.1 => /tmp/tmp/drake/lib/libvtkRenderingOSPRay-8.2.so.1 (0x00007f404d2f0000)
	libembree3.so.3 => /tmp/tmp/drake/lib/libembree3.so.3 (0x00007f404ca75000)
	libospray.so.0 => /tmp/tmp/drake/lib/libospray.so.0 (0x00007f404c845000)
	libospray_common.so.0 => /tmp/tmp/drake/lib/libospray_common.so.0 (0x00007f404c61e000)
	libospray_module_ispc.so.0 => /tmp/tmp/drake/lib/libospray_module_ispc.so.0 (0x00007f404bf90000)
	libvtkRenderingVolumeAMR-8.2.so.1 => /tmp/tmp/drake/lib/libvtkRenderingVolumeAMR-8.2.so.1 (0x00007f404bd81000)
	libvtkFiltersAMR-8.2.so.1 => /tmp/tmp/drake/lib/libvtkFiltersAMR-8.2.so.1 (0x00007f404bb5a000)
	libvtkParallelCore-8.2.so.1 => /tmp/tmp/drake/lib/libvtkParallelCore-8.2.so.1 (0x00007f404b904000)
	libvtkRenderingVolumeOpenGL2-8.2.so.1 => /tmp/tmp/drake/lib/libvtkRenderingVolumeOpenGL2-8.2.so.1 (0x00007f404b680000)
	libvtkImagingMath-8.2.so.1 => /tmp/tmp/drake/lib/libvtkImagingMath-8.2.so.1 (0x00007f404b43d000)
	libvtkRenderingVolume-8.2.so.1 => /tmp/tmp/drake/lib/libvtkRenderingVolume-8.2.so.1 (0x00007f404adb4000)
	libvtkImagingCore-8.2.so.1 => /tmp/tmp/drake/lib/libvtkImagingCore-8.2.so.1 (0x00007f404a965000)
	libvtkIOXML-8.2.so.1 => /tmp/tmp/drake/lib/libvtkIOXML-8.2.so.1 (0x00007f404a66d000)
	libvtkIOXMLParser-8.2.so.1 => /tmp/tmp/drake/lib/libvtkIOXMLParser-8.2.so.1 (0x00007f404a454000)
	libvtkRenderingSceneGraph-8.2.so.1 => /tmp/tmp/drake/lib/libvtkRenderingSceneGraph-8.2.so.1 (0x00007f404a23d000)
	libvtkIOGeometry-8.2.so.1 => /tmp/tmp/drake/lib/libvtkIOGeometry-8.2.so.1 (0x00007f4049f2a000)
	libvtkIOLegacy-8.2.so.1 => /tmp/tmp/drake/lib/libvtkIOLegacy-8.2.so.1 (0x00007f4049c73000)
	libvtkIOCore-8.2.so.1 => /tmp/tmp/drake/lib/libvtkIOCore-8.2.so.1 (0x00007f4049a0b000)
	libvtkdoubleconversion-8.2.so.1 => /tmp/tmp/drake/lib/libvtkdoubleconversion-8.2.so.1 (0x00007f40497fb000)
	libvtklzma-8.2.so.1 => /tmp/tmp/drake/lib/libvtklzma-8.2.so.1 (0x00007f40495ca000)
	libvtkRenderingOpenGL2-8.2.so.1 => /tmp/tmp/drake/lib/libvtkRenderingOpenGL2-8.2.so.1 (0x00007f40491df000)
	libvtkRenderingCore-8.2.so.1 => /tmp/tmp/drake/lib/libvtkRenderingCore-8.2.so.1 (0x00007f4048df9000)
	libvtkCommonColor-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonColor-8.2.so.1 (0x00007f4048bd8000)
	libvtkFiltersGeometry-8.2.so.1 => /tmp/tmp/drake/lib/libvtkFiltersGeometry-8.2.so.1 (0x00007f4048964000)
	libdrake_ibex.so => /tmp/tmp/drake/lib/libdrake_ibex.so (0x00007f40483cb000)
	libgfortran.so.4 => /usr/lib/x86_64-linux-gnu/libgfortran.so.4 (0x00007f4047fec000)
	libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f4047da5000)
	libvtkIOImage-8.2.so.1 => /tmp/tmp/drake/lib/libvtkIOImage-8.2.so.1 (0x00007f4047a6d000)
	libvtkDICOMParser-8.2.so.1 => /tmp/tmp/drake/lib/libvtkDICOMParser-8.2.so.1 (0x00007f4047853000)
	libvtkmetaio-8.2.so.1 => /tmp/tmp/drake/lib/libvtkmetaio-8.2.so.1 (0x00007f40475d3000)
	libvtkFiltersSources-8.2.so.1 => /tmp/tmp/drake/lib/libvtkFiltersSources-8.2.so.1 (0x00007f4047342000)
	libvtkFiltersGeneral-8.2.so.1 => /tmp/tmp/drake/lib/libvtkFiltersGeneral-8.2.so.1 (0x00007f4046e77000)
	libvtkCommonComputationalGeometry-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonComputationalGeometry-8.2.so.1 (0x00007f4046c46000)
	libvtkFiltersCore-8.2.so.1 => /tmp/tmp/drake/lib/libvtkFiltersCore-8.2.so.1 (0x00007f4046683000)
	libvtkCommonExecutionModel-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonExecutionModel-8.2.so.1 (0x00007f40463c9000)
	libvtkCommonDataModel-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonDataModel-8.2.so.1 (0x00007f4045e18000)
	libvtkCommonMisc-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonMisc-8.2.so.1 (0x00007f4045bfa000)
	libvtkCommonSystem-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonSystem-8.2.so.1 (0x00007f40459e2000)
	libvtkCommonTransforms-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonTransforms-8.2.so.1 (0x00007f40457af000)
	libvtkCommonMath-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonMath-8.2.so.1 (0x00007f404558d000)
	libvtkCommonCore-8.2.so.1 => /tmp/tmp/drake/lib/libvtkCommonCore-8.2.so.1 (0x00007f4044ffe000)
	libvtksys-8.2.so.1 => /tmp/tmp/drake/lib/libvtksys-8.2.so.1 (0x00007f4044dbf000)
	libyaml-cpp.so.0.5 => /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.5 (0x00007f4044b45000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f4044913000)
	libjsoncpp.so.1 => /usr/lib/x86_64-linux-gnu/libjsoncpp.so.1 (0x00007f40446e1000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f40444c5000)
[... cut ...]
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f403e680000)
	libpord_seq-5.1.2.so => /usr/lib/x86_64-linux-gnu/libpord_seq-5.1.2.so (0x00007f403e46a000)
	libesmumps-6.so => /usr/lib/x86_64-linux-gnu/libesmumps-6.so (0x00007f403e265000)
	libscotch-6.so => /usr/lib/x86_64-linux-gnu/libscotch-6.so (0x00007f403dfd5000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f403ddce000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f403dbb9000)
	libscotcherr-6.so => /usr/lib/x86_64-linux-gnu/libscotcherr-6.so (0x00007f403d9b7000)
```

BetsyMcPhail, 2020-08-29:
> ./cart_pole/cart_pole_passive_simulation

I followed this script and confirmed that it works on my system.

Maverobot, 2020-08-29:
> > ./cart_pole/cart_pole_passive_simulation
> 
> I followed this script and confirmed that it works on my system.

Strange. Can we guess with the error message what could go wrong on my setup?

jwnimmer-tri, 2020-09-03:
> clang version 10.0.0-4ubuntu1~18.04.2 

Oh, I just noticed this.  You aren't using clang to compile your example, are you?  (Is CC or CXX set to clang?)

Maverobot, 2020-09-03:
> > clang version 10.0.0-4ubuntu1~18.04.2
> 
> Oh, I just noticed this. You aren't using clang to compile your example, are you? (Is CC or CXX set to clang?)
> 
> You should be using GCC.

Yes I was using Clang. And It actually does not matter since with GCC I still get the same error

```fish
terminate called after throwing an instance of 'std::logic_error'
  what():  Can't acquire pose port for unknown source id: 93928621898016.
fish: “./cart_pole/cart_pole_passive_s…” terminated by signal SIGABRT (Abort)
```

jwnimmer-tri, 2020-09-03:
> ... with GCC I still get the same error ...

Can you re-confirm this is true with a completely clean / fresh image with no sign of clang-compiled code anywhere?

Maverobot, 2020-09-03:
I finally got it working with `gcc-7`. Unfortunately it throws error with `gcc-9` and `clang`.

jwnimmer-tri, 2020-09-03:
Yes, Drake's pre-compiled binary releases use the platform's default compiler.  For Ubuntu 18.04, that is gcc-7.  You cannot mix and match compilers, in general.

closed 2020-09-03 by jwnimmer-tri

Maverobot, 2020-09-03:
You are right. I know many people like using higher version of gcc to support latest C++ standard e.g. c++20. Maybe it is a good idea to mention in the documentation that `gcc-7` is the required compiler if the pre-compiled drake library is used.