RobotLocomotion/drake #13934: Improve MeshcatVisualizer frames_to_draw API

RussTedrake, 2020-08-26:
I'm not sure how the current API got through code review?  We should absolutely *not* pass the **names** of the model instances / frames as strings to this method as a dictionary.  The [current test](https://github.com/RobotLocomotion/drake/blob/2b44dd1325e42bf9b51b139dd2594bf93672d629/bindings/pydrake/systems/test/meshcat_visualizer_test.py#L115) already demonstrates one reason why -- you have to check that the frames exist with a second call?  But you are also forcing an unnecessary dependency on `MultibodyPlant` for something that should work for any geometry frame known by `SceneGraph`.  

I think there is no doubt that the right API here is to take a list of `geometry::FrameId`'s.  ?

I think we need to deprecate the old API.  I would also say that the argument name is a bit misleading.  I would prefer `triads_to_draw`, perhaps?  The current name suggests that geometry associated with frames not on the whitelist would not be drawn?  So the deprecation strategy I would propose is to add the deprecation warning to the current argument and add an entirely new argument that implements the frameId version?

@pangtao22 -- this was your feature.  Can you take a pass at updating it?
cc @EricCousineau-TRI (co-author on the commit)

BTW -- it's a nice feature!  and I'm super glad it exists (I'm using it now).  Just want it to be nicer.

EricCousineau-TRI, 2020-08-26:
I think a good first pass would be to rewrite `MeshcatVisualizer` to work off of `QueryObject` (#10482).
Then maybe the weird hacks for model instance could be removed?

pangtao22, 2020-08-27:
The nice thing about using names is that they can be read (by humans) from the SDF files before constructing anything in drake. In contrast, `geometry::FrameId` is a list of integers which are less intuitive to interpret, and you may not know what they are until the `MultibodyPlant` is finalized? 

When the names passed in do not exist, `MeshcatVisualizer` simply ignore those names without giving warnings, no checks that depend on `MultibodyPlant` are performed:https://github.com/RobotLocomotion/drake/blob/2b44dd1325e42bf9b51b139dd2594bf93672d629/bindings/pydrake/systems/meshcat_visualizer.py#L409-L410. 

It is therefore the user's duty to type in the correct names, which I think is reasonable. The reason why their existence is checked in the unit test is to make sure that the code to draw frames is not bypassed due to wrong frame names. 

I'd love to update the API to something that's equally intuitive and up-to-date with drake's latest best practices.

RussTedrake, 2020-08-31:
I would fully expect a user to still be able to use string names, but it would be via e.g. `plant.GetFrameByName()` or `plant.GetBodyByName()` and friends.  I think `MeshcatVisualizer` should fail if the frames don't exist.  And I think the MBP checks should happen before calling the function (i agree that there should not be a dependency on MBP).

having said that, it's possible that we don't have a sufficient API on `SceneGraph` yet to make the api that I'm asking for a reality.  That might be what @EricCousineau-TRI meant by bringing up `QueryObject`.  But it's worth trying, no?

SeanCurtis-TRI, 2020-08-31:
FTR Given a frame id, you can always ask the `SceneGraphInspector` for its name. `MBP` cobbles together a frame name based on model instance and body name.

RussTedrake, 2021-05-04:
will close this based on #14979

closed 2021-05-04 by RussTedrake