RobotLocomotion/drake #13945: pydrake: Should add (canary?) testing for running under anaconda

EricCousineau-TRI, 2020-08-27:
Relates #13944

See Slack convo:
https://github.com/RobotLocomotion/drake/issues/13944

Possibly add a minor test in `drake-external-examples` consuming the binaries.

jwnimmer-tri, 2020-08-27:
We should either fully support anaconda, or not give any impression that we do; there is no middle ground.  So the question is not whether we should add a canary or examples, but whether we should choose to support this installation method and runtime environment at all.

EricCousineau-TRI, 2020-08-27:
Updated title.

BenBurchfiel-TRI, 2020-09-23:
> We should either fully support anaconda, or not give any impression that we do; there is no middle ground. So the question is not whether we should add a canary or examples, but whether we should choose to support this installation method and runtime environment at all.

What about supporting just conda? We could support conda installing drake into a new environment, but not into a full anaconda environment. This would let both miniconda and anaconda trivially install Drake but still ensure that there wasn't a bunch of other junk conflicting with it.

jwnimmer-tri, 2020-09-24:
My point was more about the formulation -- "add testing for an unsupported platform" (per the original issue title) is not my preference.

I'm not well informed on the difficulty of doing "just conda" or the related ecosystem.  Someone who is interested and informed could use this issue to discuss.

The difficulties with pip (#1183) related to the C++ side of the dependencies.  It would be interesting to learn what that story was for conda.

EricCousineau-TRI, 2021-01-22:
Merp :(
https://github.com/conda/conda/issues/10470

EricCousineau-TRI, 2021-01-22:
Note that the installation docs for Conda are part of the reason why it may not readily play well with stuff like Drake (non-platform supplied packages, invasive use of user environment, etc.):
https://github.com/conda/conda/issues/10470

We could also actively advise people to use "silent mode" (per the conda docs) as means to avoid intermixing.

jwnimmer-tri, 2023-05-22:
I'm closing this feature request as "not planned".  If someone wants to do the work, we can consider accepting it and supporting conda, but at the moment the core build & release team cannot support this.