RobotLocomotion/drake #13950: CalcJacobianSpatialVelocity documentation render badly on mac

RussTedrake, 2020-08-27:
[image]

It seems that our previous font gymnastics did not work for this one?  (some of the other jacobian methods are borked, too)

sherm1, 2020-08-27:
@jamiesnape do you have any more magic tricks up your sleeve?

mitiguy, 2020-08-27:
Tomorrow, I'll take a look to see if I used any strange fonts.

jwnimmer-tri, 2020-08-27:
FYI #10628 was the PR that resolved this last time around.

@mitiguy It's not that you've chosen a weird font, it's whether or not the macOS DejaVu Sans contains the characters used in that text (some kind of slanted S, it looks like?). 

See https://github.com/RobotLocomotion/drake/blob/6bd91f5a5bf3855ea91b794b29e1c22a895b65a0/doc/doxygen_extra.css where we choose which font to use.

@sherm1 Do you still have your mac to test on?  Are you able to reproduce the problem?

\CC @soonho-tri

jwnimmer-tri, 2020-08-27:
FYI Here's what it looks like on Ubuntu Firefox

[image]

jwnimmer-tri, 2020-08-27:
@RussTedrake to confirm -- on your mac, you're definitely run install_prereqs?  In particular:

https://github.com/RobotLocomotion/drake/blob/6bd91f5a5bf3855ea91b794b29e1c22a895b65a0/setup/mac/source_distribution/Brewfile#L4-L7

sherm1, 2020-08-28:
Thanks @jwnimmer-tri. I don't currently have access to a Mac (still in a drawer at HQ unless Sean has it at home).
However, a solution that occurs to me would be to forgo the slant in the slanted-s and just use a plain s there. What do you think about that @mitiguy ? We can then have @RussTedrake feature review it on his Mac.

jwnimmer-tri, 2020-08-28:
It wasn't just the s -- the `$\triangleq$` also looks broken, as does the n subscript, etc.

soonho-tri, 2020-08-28:
It looks OK to me on my local mac.

[image]

https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a5f39a8f68fae3de53767a281503a3313

mitiguy, 2020-08-28:
Maybe this is already known -- but looking at Russ's original post, it seems this font problem only occurs in situations involving <pre> such as:

<pre>
   Some equations here.
</pre>

RussTedrake, 2020-08-28:
@jwnimmer-tri -- i run the install script often.  Just ran it again... excerpt:
```
++ /usr/local/bin/brew bundle --file=./setup/mac/source_distribution/Brewfile --no-lock
Updating Homebrew...
Using homebrew/cask-fonts
Using robotlocomotion/director
Using font-dejavu
Using bazel
```
I restarted chrome just to be sure.  Still see garbage.

@soonho-tri what browser are you using?

RussTedrake, 2020-08-28:
also, it hadn't actually occurred to me that for many users:
google colab => you can run drake with no install => you will not be able to read all of our documentation (on mac).

soonho-tri, 2020-08-28:
@RussTedrake , 
 - Chrome-85.0.4183.83 and Firefox-80.0.0 : Works fine.
 - Safari-13.1.2 : does not render the subscript n in `s_n` correctly. PTAL the following screenshot.

[image]

The general advice is to install the four font-families mentioned in https://drake.mit.edu/unicode_tips_tricks.html. Could you check you have all of them installed in your system?

BTW, those links for Noto fonts are not working. Please use the following:

 - https://www.google.com/get/noto/#sans-lgc
 - https://www.google.com/get/noto/#sans-zsym

jwnimmer-tri, 2020-08-28:
This is more of a distribution & release issue than a dynamics issue.  Users need to have the right fonts to use our docs, beyond just the jacobians methods.  Maybe we can do better about facilitating that.

mitiguy, 2020-08-28:
@jwnimmer-tri  - I agree and thanks for looking at this.  I am ill-suited to handle a general fonts distribution issue.

RussTedrake, 2020-08-28:
@mitiguy -- I wasn't asking you to handle a general fonts distribution issue.  Consider adding the latex alongside (potentially even instead of) your unicode for these methods?  I think that the argument in favor of unicode readability in code editors needs to be balanced with readability in doxygen.  

In this particular case, it is to the point that my render in doxygen made it almost impossible for me (as a relative expert) to know what the arguments were referring to.  The information is encoded with the frame names A, B, and E, and I felt like I wasn't given the secret decoder ring.

jamiesnape, 2020-08-28:
For the online docs, you shouldn't actually need the fonts installing on any platform. The browser should download as needed. So do we think this problem is unique to `<pre>`, i.e., monospace fonts?

jwnimmer-tri, 2020-08-28:
@jamiesnape None of us can reproduce it, so its hard to guess what's going on.  The only sensible way to fix this is by someone who can reproduce it.  So far, it looks fine for Soonho, but not for Russ.  What does it look like on your mac(s)?

jwnimmer-tri, 2020-08-28:
(I checked safari on iPhone, and it looks fine.)

jamiesnape, 2020-08-28:
Looks the same as Soonho, one missing subscript.

jwnimmer-tri, 2020-08-28:
@RussTedrake which browser(s) versions are you seeing the failure in?

RussTedrake, 2020-08-28:
I do recall one former installation that was failing on install_prereqs updating font-dejavu, and the advice I was given was to hard remove the previously installed dejavu fonts.  Perhaps the problem dates back to that action?

It *does* view fine on Safari
It fails on my default browser, Mac Chrome Version 84.0.4147.135 (Official Build) (64-bit).

SeanCurtis-TRI, 2020-08-28:
I tried it safari on an ipad, and one render problem: the subscript n renders as a box.

RussTedrake, 2020-08-28:
I tried Chrome on ipad, and see the same thing as Sean.  I also just tried full rebooting my mac, but the problem on macbook Chrome persists.

jamiesnape, 2020-08-28:
For the `pre` block, looks like there is an issue with the cascade. I am seeing the rendering as the generic `monospace` font family. Looking at the stylesheet, that is actually the correct (though undesirable) behavior. The sledgehammer is to place `pre` here: https://github.com/RobotLocomotion/drake/blob/6bd91f5a5bf3855ea91b794b29e1c22a895b65a0/doc/doxygen_extra.css#L4

jamiesnape, 2020-08-28:
So not a Mac issue per se, we just hit a style combination that Doxygen was not expecting. Difficult to say there are not others elsewhere without looking much more deeply. https://github.com/RobotLocomotion/drake/commit/07afd0e86d7c23438fed35580747391254c9fca0 was actually a portent regarding the style of the `pre` class.

jamiesnape, 2020-08-28:
We are missing overrides for these:
* https://chromium.googlesource.com/chromium/blink/+/master/Source/core/css/html.css#930 (Chrome)
* https://dxr.mozilla.org/mozilla-central/source/layout/style/res/html.css#244-259 (Firefox)
* https://trac.webkit.org/browser/trunk/Source/WebCore/css/html.css#L1077 (Safari)

jamiesnape, 2020-08-28:
https://github.com/RobotLocomotion/drake/pull/13952

mitiguy, 2020-08-28:
@RussTedrake   To help me better understand the problem, would it be possible for you to check if the font issue seems to be most problematic when a  \<pre\> tag is involved.  Does the problem mostly/completely disappear when you remove the surrounding \<pre\> tags?   If so, perhaps we could consider additional solutions (in addition to Latex).

jwnimmer-tri, 2020-08-28:
@mitiguy Jamie already solved this (see the linked PR).  The doxygen infrastructure should support the kinds of comments that developers want to write.  It had a bug where it failed to apply a font choice.  You shouldn't need to worry about the docs you write. If they render correctly for you in your own testing, its the job of our infrastructure to ensure that works for others as well.

PR #13952 (2020-08-31): Explicitly set font-family for all monospaced HTML elements
Closes #13950.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13952)
closed 2020-08-31 by soonho-tri