RobotLocomotion/drake #13964: doxygen on mac mysteriously skips MultibodyPlant Jacobian methods

RussTedrake, 2020-08-29:
As discussed in #13952, after running `bazel-bin/doc/doxygen` on mac, the methods `CalcJacobianSpatialVelocity` and friends are simply missing.
[image]

soonho-tri, 2020-08-29:
FYI, I think it has something to do with the use of `DRAKE_DEPRECATED` macro. When I removed them (there are three occurrences), it rendered those methods.

jamiesnape, 2020-09-02:
Everything is there for me with Doxygen 1.8.20 on Catalina. That only became available less than 7 days ago. I will check 1.8.19.

BetsyMcPhail, 2020-09-14:
Thanks for the hint @soonho-tri, I'm able to reproduce on my system now. If I build the docs _before_  #13939 (_Remove deprecated code (2020-09)_), `CalcJacobianSpatialVelocity` and friends are missing. After that commit, they are included.

BetsyMcPhail, 2020-09-17:
It turns out all macros, not just ``DRAKE_DEPRECATED``, aren't being handled correctly on Mac. For example:

Mac generated documentation:

[image]

From https://drake.mit.edu/doxygen_cxx/classdrake_1_1symbolic_1_1_expression_add_factory.html

[image]

Downgrading from Doxygen 1.8.20 to 1.8.15 solved the issue on my system.

jwnimmer-tri, 2020-09-17:
@BetsyMcPhail can you try to find more details about the regression?  For example, check if `PREDEFINED` meaning or spelling changed, or if ...

```
ENABLE_PREPROCESSING   = YES
MACRO_EXPANSION        = YES
EXPAND_ONLY_PREDEF     = YES
```

... are somehow different, or if the quoting rules changed, or etc?  Hopefully we can resolve this through a configuration edit, or if not be able to report a concise bug upstream about what broke.

jwnimmer-tri, 2020-09-17:
To be clear though, I am more than happy to have us unify on a single version of Doxygen on all platforms -- I just also want to look into the deeper problem.

jwnimmer-tri, 2020-09-17:
FYI https://github.com/RobotLocomotion/drake/pull/13266#pullrequestreview-410132243 might be related (that the macros stopped working).  It mentions that 1.8.17 was a known-broken revision.

jwnimmer-tri, 2020-10-01:
Update from f2f: Jamie compiled an older revision of Doxygen and staged into mac CI with latest images, but it's not in production use yet.  We're going to work on switching macOS to use that in the interim until we can diagnose the upstream issues.

BetsyMcPhail, 2020-10-09:
I confirmed that 1.8.16 parses macros as expected and 1.8.17 shows the errors.

BetsyMcPhail, 2020-10-09:
git bisect tells me that commit [dbd8ceb8e](https://github.com/doxygen/doxygen/commit/dbd8ceb8efa2e7b413780a43e74b30f039670a72) is the origin of the problem.

Doxygen issues [7527](https://github.com/doxygen/doxygen/issues/7527) and [7509](https://github.com/doxygen/doxygen/issues/7509) seem to describe the issue we're seeing.

BetsyMcPhail, 2020-10-09:
Also confirmed that [commit 938b6dfd](https://github.com/doxygen/doxygen/commit/938b6dfd1cb5173aa6ab19aa848ae266dba0afff) (proposed fix for issue 7527) did not fix the issue.

jwnimmer-tri, 2020-10-10:
In terms of work-arounds, perhaps our `doc/doxygen_main.py` could call into https://github.com/ned14/pcpp to pre-process the `DRAKE_DEPRECATED` (and the other PREDEFINED macros) and rewrite all of the files, so that we don't have to rely on Doxygen's broken pre-processing.

jamiesnape, 2020-10-14:
I think we are going to get a minimal repro and file a bug with doxygen first. If we can get a fix in easily too, that might be a little easier than adding a different preprocessor, since we this is only a tool and we already build packages for two of three platforms. @BetsyMcPhail will post when we have an update from upstream.

BetsyMcPhail, 2020-10-20:
Filed Doxygen issue [#8119 ](https://github.com/doxygen/doxygen/issues/8119)and PR [#8120](https://github.com/doxygen/doxygen/pull/8120)

jwnimmer-tri, 2020-10-20:
Super!

BetsyMcPhail, 2020-10-20:
Fix has been merged into Doxygen master.

BetsyMcPhail, 2020-10-20:
Yes, I think we can call this closed

closed 2020-10-20 by BetsyMcPhail