RobotLocomotion/drake #13970: binding request: set visual geometry alpha from python

RussTedrake, 2020-08-30:
I was hoping to try the workflow from @mposa to change the transparency of some geometries after parsing.  
https://github.com/DAIRLab/dairlib/blob/master/multibody/multipose_visualizer.cc#L59  
The catch is that I want to do it from python. 

Here is a quick prototype of a python example which exposed the missing bindings for me.
```
from pydrake.all import (
    AddMultibodyPlantSceneGraph, DiagramBuilder, ConnectMeshcatVisualizer, 
    FindResourceOrThrow, Parser)

builder = DiagramBuilder()

plant, scene_graph = AddMultibodyPlantSceneGraph(builder, time_step = 0.0)
parser = Parser(plant, scene_graph)
grasp = parser.AddModelFromFile(FindResourceOrThrow(
    "drake/manipulation/models/wsg_50_description/sdf/schunk_wsg_50.sdf"), "grasp")
pregrasp = parser.AddModelFromFile(FindResourceOrThrow(
    "drake/manipulation/models/wsg_50_description/sdf/schunk_wsg_50.sdf"), "pregrasp")
brick = parser.AddModelFromFile(FindResourceOrThrow(
    "drake/examples/manipulation_station/models/061_foam_brick.sdf"))
plant.Finalize()

def set_transparency(plant, scene_graph, model_instance, alpha):
    inspector = scene_graph.model_inspector()
    body_indices = plant.GetBodyIndices(model_instance)
    for body_index in body_indices:
        geometry_ids = plant.GetVisualGeometriesForBody(plant.get_body(body_index))
        for geometry_id in geometry_ids:
            prop = inspector.GetIllustrationProperties(geometry_id)
            new_prop = IllustrationProperties(prop)
            phong = prop.GetProperty("phong", "diffuse")
            phong.set(phong.r(), phong.g(), phong.b(), new_alpha);
            new_props.UpdateProperty("phong", "diffuse", phong);
            scene_graph.AssignRole(plant.get_source_id().value(), geometry_id,
                                   new_props, RoleAssign.kReplace)
          
set_transparency(plant, scene_graph, pregrasp, 0.7)

diagram = builder.Build()
context = diagram.CreateDefaultContext()
```

I suspect there are issues with it, but the first one is 
```
AttributeError: 'MultibodyPlant_[float]' object has no attribute 'GetVisualGeometriesForBody'
```
I would love to add the few bindings required to make this workflow work (and am happy to follow a different workflow if you recommend something different).

@SeanCurtis-TRI -- feel free to delegate.

SeanCurtis-TRI, 2020-09-02:
We want to stay away from the MBP API that knows about the geometry ids. That's an anachronism that is to be excised soon. The alternate spelling would be:

```py
from pydrake.all import (
    AddMultibodyPlantSceneGraph, DiagramBuilder, ConnectMeshcatVisualizer, 
    FindResourceOrThrow, Parser)

builder = DiagramBuilder()

plant, scene_graph = AddMultibodyPlantSceneGraph(builder, time_step = 0.0)
parser = Parser(plant, scene_graph)
grasp = parser.AddModelFromFile(FindResourceOrThrow(
    "drake/manipulation/models/wsg_50_description/sdf/schunk_wsg_50.sdf"), "grasp")
pregrasp = parser.AddModelFromFile(FindResourceOrThrow(
    "drake/manipulation/models/wsg_50_description/sdf/schunk_wsg_50.sdf"), "pregrasp")
brick = parser.AddModelFromFile(FindResourceOrThrow(
    "drake/examples/manipulation_station/models/061_foam_brick.sdf"))
plant.Finalize()

def set_transparency(plant, scene_graph, model_instance, alpha):
    inspector = scene_graph.model_inspector()
    body_indices = plant.GetBodyIndices(model_instance)
    for body_index in body_indices:
        frame_id = plant.GetBodyFrameIdOrThrow(body_index)
        geometry_ids = inspector.GetGeometries(frame_id, Role.kIllustration)  ## THIS LINE!!!
        for geometry_id in geometry_ids:
            prop = inspector.GetIllustrationProperties(geometry_id)
            new_prop = IllustrationProperties(prop)
            phong = prop.GetProperty("phong", "diffuse")
            phong.set(phong.r(), phong.g(), phong.b(), new_alpha);
            new_props.UpdateProperty("phong", "diffuse", phong);
            scene_graph.AssignRole(plant.get_source_id().value(), geometry_id,
                                   new_props, RoleAssign.kReplace)
          
set_transparency(plant, scene_graph, pregrasp, 0.7)

diagram = builder.Build()
context = diagram.CreateDefaultContext()
```

In that code, the line marked `## THIS LINE!!!` is the missing piece. It has been oft discussed and its absence frequently mourned. I'd much rather use this as the catalyst to finally add the missing API.

SeanCurtis-TRI, 2020-09-03:
As noted [in this code review](https://github.com/RobotLocomotion/drake/pull/13999#pullrequestreview-481097010), the proposed code doesn't work as is. There are no bound copy-constructors for geometry properties. So,

```py
new_prop = IllustrationProperties(prop)
```

Will not work.

There are a couple of options:
  - [https://github.com/RobotLocomotion/drake/pull/13999#pullrequestreview-481272629](Enabling copy/deep copy on the classes)
  - [https://github.com/RobotLocomotion/drake/pull/13999#pullrequestreview-481272909](Building it into the class)

To fully enable the proposed workflow, we need this as well.

SeanCurtis-TRI, 2020-09-03:
Note: #14010 fills the indicated hole, enabling:

```py
new_prop = IllustrationProperties(prop)
```

PR #14010 (2020-09-03): [pydrake] Enable GeometryProperty copy construction in pydrake
The {Proximity|Illustration|Perception}Properties now have a copy constructor in pydrake. This facilitates the ability to update/replace properties.

This is the second piece necessary to meaningfully enable #13970.

resolves #13970.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14010)
closed 2020-09-03 by SeanCurtis-TRI