RobotLocomotion/drake #13973: trajectories: refactor `PiecewiseCartesianTrajectory` proposal

RussTedrake, 2020-08-30:
I just found `PiecewiseCartesianTrajectory` hiding in `manipulation/util/trajectory_utils.h`.  But everything it needs is in `common\trajectories` -- most notably the `PiecewiseQuaternionSlerp`.

Also hiding next to it is `PiecewiseCubicTrajectory`, which I *think* should not exist.  `PiecewisePolynomial` should play that role.  If there is something missing, we should add it there?

Proposal: Can we relocate `PiecewiseCartesianTrajectory` (is that the right name?) to `common\trajectories` and deprecate `PiecewiseCubicTrajectory`?

@siyuanfeng-tri -- wdyt?

of course we will deprecate.  but someone considers this a major dependency, please let me know.

anyone looking for a good deed to do?

RussTedrake, 2020-08-30:
If I were to rename it, I would think `PiecewisePoseTrajectory` would be the natural choice?

siyuanfeng-tri, 2020-08-31:
> If I were to rename it, I would think PiecewisePoseTrajectory would be the natural choice?

I think that'd be fine. I am on the fence for being more specific. It is a very generic name, and probably should be reserved for a base class. 

> Also hiding next to it is PiecewiseCubicTrajectory, which I think should not exist.

You are probably right. The reason I had it in the first place is to factorize a bit better with Cartesian trajectories that have different rotation parts. For most usages in anzu, we almost never do multi-piece move tool, so it's a bit nicer to have a smoother orientation part for just one piece.

PR #14337 (2021-01-06): trajectories: Refactor PiecewiseCartesianTrajectory to PiecewisePoseTrajectory
Also deprecates `PiecewiseCartesianTrajectory` and `PiecewiseCubicTrajectory`

Closes #13973

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14337)
closed 2021-01-06 by RussTedrake