RobotLocomotion/drake #13975: plot_graphviz / GetTopologyGraphvizString generates too-small-to-read figures

ggould-tri, 2020-08-31:
This notebook https://colab.research.google.com/github/RussTedrake/manipulation/blob/master/pick.ipynb#scrollTo=wSa5QcU0Piak in @RussTedrake 's course materials makes this stylish little kinematic tree graph:
[image]
Emphasis on "little" -- that image is 350x218 pixels, and many of the letters are a bare eight pixels tall.  There's no obvious way to make the notebook make it bigger, and even zoomed it is hard to read.

Graphviz scaling is a pain, but I bet @rpoyner-tri knows what the right config is, either in generating the graphviz figure or in invoking the plot command.

(Assigning this to dynamics because the code seems to be in their area)

RussTedrake, 2020-08-31:
Thanks Grant!  But I think the *right* solution is to render with javascript -- not png.  Would love help with that!

ggould-tri, 2020-08-31:
Or svg perhaps, for us static content dead-enders :-) .  Labeling this as low-priority in case #13874 arrives sooner.

sherm1, 2020-09-02:
Assigning to @rpoyner-tri to see if Grant is right.

jwnimmer-tri, 2020-09-02:
For example, https://dagrejs.github.io/project/dagre-d3/latest/demo/interactive-demo.html has click-to-pan and scrollwheel-to-zoom.

RussTedrake, 2020-09-06:
I think SVG goes a long way for now.  I've updated my notebook to use:
```
from IPython.display import SVG
import pydot
SVG(pydot.graph_from_dot_data(plant.GetTopologyGraphvizString())[0].create_svg())
```

I'll mark this as resolved for now, but would love to continue improving this!

PS - @jwnimmer -- I've been looking at dagre.js, too..  I would have started using it for the diagram workflow, if it only had collapsable subdiagrams!

closed 2020-09-06 by RussTedrake