RobotLocomotion/drake #13985: move-aware implementation of drake's Eigen::AutoDiffScalar<VectorXd>

rpoyner-tri, 2020-09-01:
Early experiments have shown that modernizing methods and math operations for drake's Eigen::AutoDiffScalar<VectorXd> can reduce heap allocations and provide some performance benefit. (I claimed elsewhere that 2x speedup was possible; I'm no longer sure that is available from this group of changes alone.)

This ticket is to track conversion of AutodiffScalar to move-awareness. I will also provide performance measurements as the work proceeds.

Related issue for overall autodiff speedup: #10991.

rpoyner-tri, 2020-09-01:
Here are some early results. Comparison is from master at 1f51d66cb2f20b1248fe216f17943241a53a2797 to this tag in my fork: https://github.com/rpoyner-tri/drake/releases/tag/move-dev-benchmark0 .

Both experiments are done on my Puget workstation, using commands like this:
`$ examples/multibody/cassie_benchmark/conduct_experiment [DIRECTORY] --benchmark_repetitions=100`

Results from master:
[outputs.zip](https://github.com/RobotLocomotion/drake/files/5157302/outputs.zip)

Some selected numbers (min and stddev, with allocation statistics as well) from master:
```
CassieDoubleFixture/DoubleMassMatrix_stddev                 120 ns          120 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieDoubleFixture/DoubleMassMatrix_min                  16527 ns        16526 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieDoubleFixture/DoubleInverseDynamics_stddev            150 ns          150 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieDoubleFixture/DoubleInverseDynamics_min             23003 ns        23002 ns          100 Allocs.max=3 Allocs.mean=3 Allocs.min=3 Allocs.stddev=0
CassieDoubleFixture/DoubleForwardDynamics_stddev            294 ns          294 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieDoubleFixture/DoubleForwardDynamics_min             45397 ns        45396 ns          100 Allocs.max=22 Allocs.mean=22 Allocs.min=22 Allocs.stddev=0
CassieAutodiffFixture/AutodiffMassMatrix_stddev           46706 ns        46701 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieAutodiffFixture/AutodiffMassMatrix_min            4482277 ns      4482121 ns          100 Allocs.max=54.465k Allocs.mean=54.465k Allocs.min=54.465k Allocs.stddev=0
CassieAutodiffFixture/AutodiffInverseDynamics_stddev     109909 ns       109907 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieAutodiffFixture/AutodiffInverseDynamics_min       5983179 ns      5982889 ns          100 Allocs.max=68.798k Allocs.mean=68.798k Allocs.min=68.798k Allocs.stddev=0
CassieAutodiffFixture/AutodiffForwardDynamics_stddev     131372 ns       131362 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieAutodiffFixture/AutodiffForwardDynamics_min       8887333 ns      8886994 ns          100 Allocs.max=102.904k Allocs.mean=102.904k Allocs.min=102.904k Allocs.stddev=0
```

Here are results from the development tag given above.
[outputs.zip](https://github.com/RobotLocomotion/drake/files/5157372/outputs.zip)

Some summary numbers show improvement in autodiff benchmarks, and perhaps a slight penalty in double benchmarks.
```
CassieDoubleFixture/DoubleMassMatrix_stddev                81.4 ns         81.4 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieDoubleFixture/DoubleMassMatrix_min                  16578 ns        16578 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieDoubleFixture/DoubleInverseDynamics_stddev            265 ns          265 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieDoubleFixture/DoubleInverseDynamics_min             23087 ns        23086 ns          100 Allocs.max=3 Allocs.mean=3 Allocs.min=3 Allocs.stddev=0
CassieDoubleFixture/DoubleForwardDynamics_stddev            338 ns          338 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieDoubleFixture/DoubleForwardDynamics_min             45622 ns        45621 ns          100 Allocs.max=22 Allocs.mean=22 Allocs.min=22 Allocs.stddev=0
CassieAutodiffFixture/AutodiffMassMatrix_stddev           30190 ns        30196 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieAutodiffFixture/AutodiffMassMatrix_min            3321523 ns      3321480 ns          100 Allocs.max=35.043k Allocs.mean=35.043k Allocs.min=35.043k Allocs.stddev=0
CassieAutodiffFixture/AutodiffInverseDynamics_stddev      32186 ns        32184 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieAutodiffFixture/AutodiffInverseDynamics_min       4503590 ns      4503402 ns          100 Allocs.max=41.644k Allocs.mean=41.644k Allocs.min=41.644k Allocs.stddev=0
CassieAutodiffFixture/AutodiffForwardDynamics_stddev     114352 ns       114355 ns          100 Allocs.max=0 Allocs.mean=0 Allocs.min=0 Allocs.stddev=0
CassieAutodiffFixture/AutodiffForwardDynamics_min       6345443 ns      6345197 ns          100 Allocs.max=61.376k Allocs.mean=61.376k Allocs.min=61.376k Allocs.stddev=0
```

The branch has achieved a rough 40% reduction in heap allocations for autodiff cases, perhaps 25% speed-up of autodiff benchmark timings, and maybe a slighty penalty for double cases.

Let's use the [googlebench comparison tool](https://github.com/google/benchmark/blob/master/docs/tools.md#comparepy) to see how the two data sets compare. The full comparison report is attached in comparison.zip. The report is ansi-colored text; use `less -R` or similar to view it.
[comparison.zip](https://github.com/RobotLocomotion/drake/files/5157538/comparison.zip)

Rather than struggle with the wall of numbers that is the full report, let's just look at `_min` (@sherm1 's favorite number), and the U test p-value, which measures something like (inverse?) likelihood that the two data sets actually differ. 0 means high likelihood of difference; values larger mean less likelihood of difference.
[... cut ...] m/RobotLocomotion/drake/files/5157574/comparison_summary.html.pdf)
(I converted ansi-colored text to html then pdf to present colors here at all. For most numbers: red=bad, gray=meh, blue=good. For p-values: red=bad (above threshold), green=good.)

From the summary, we can see that autodiff cases got roughly 25% percent faster, with high confidence. The double mass matrix case remained pretty much the same, and the double forward and inverse dynamics cases got a hair worse, with high confidence.

sherm1, 2020-09-01:
Great that the AutoDiff cases are significantly faster! But I'm mystified by the report on the other cases. Since they don't use AutoDiff at all, why were inverse & forward affected "with high confidence"? Also, the p-value for mass matrix was huge, yet the runtime changes were similar to inverse and forward whose p-values were 0! Those 0 p-values seem absurd to me.

rpoyner-tri, 2020-09-01:
@sherm1 You are right -- the p-values for the double cases probably deserve more introspection. The various attached zip files have all the relevant raw data, so I could poke around in re-runs of the compare.py script to see what is going on. I do know that the script uses the Mann-Whitney implementation from `scipy`, so I probably am justified in trusting that. Not sure about other aspects of how the data are handled though.

rpoyner-tri, 2020-10-05:
I'm reasonably certain that all the low-hanging fruit from this strategy has been harvested. There might be more at the margins, but see #14039 for obstacles. Closing.

closed 2020-10-05 by rpoyner-tri