RobotLocomotion/drake #13987: pydrake doc's default arguments have memory hex addresses

jwnimmer-tri, 2020-09-01:
A few places in our generated docs have hex memory addresses from repr embedded in the API signature:
- [FiniteHorizonLinearQuadraticRegulator](https://drake.mit.edu/pydrake/pydrake.systems.controllers.html#pydrake.systems.controllers.FiniteHorizonLinearQuadraticRegulator)
- [RgbdSensor](https://drake.mit.edu/pydrake/pydrake.systems.sensors.html?r#pydrake.systems.sensors.RgbdSensor.__init__)
- [RegionOfAttraction](https://drake.mit.edu/pydrake/pydrake.systems.analysis.html#pydrake.systems.analysis.RegionOfAttraction)
- etc.

It's usually from pybound C++ structs that don't have good `__str__` or `__repr__`.

We should make the objects print correctly, so the docs don't have line noise.  (The line noise churns out deployment diffs, also.)

EricCousineau-TRI, 2020-09-01:
Aye, thanks for the filing! Simplest is to see if there is a way to somehow express them in a `dataclass` type way. Then we can get `repr()` that is something like `MyStruct(my_param=...)`.

jwnimmer-tri, 2021-04-28:
These appear to be the current set with hex addresses in the docs:
```
pydrake.systems.framework.UseDefaultNamepydrake.geometry.DrakeVisualizerParams
pydrake.perception.Fields
pydrake.systems.analysis.InitializeParams
pydrake.systems.analysis.RegionOfAttractionOptions
pydrake.systems.controllers.FiniteHorizonLinearQuadraticRegulatorOptions
pydrake.systems.framework.UseDefaultName
```

jwnimmer-tri, 2021-05-02:
Huh.  Somehow I missed that `RigidTransform_` is missing a repr.

jwnimmer-tri, 2023-01-04:
We have some new ones:
- `pydrake.math.NumericalGradientOption`
- `pydrake.geometry.render.RenderLabel`
- `pydrake.solvers.SolverOptions`

jwnimmer-tri, 2024-06-03:
Two more:
- `Parallelism`
- `SpatialInertia`

PR #14959 (2021-04-28): pydrake: Add `__repr__` to several classes
Specifically, add `__repr__` for:
- pydrake.geometry.DrakeVisualizerParams
- pydrake.perception.Fields
- pydrake.systems.analysis.InitializeParams
- pydrake.systems.analysis.RegionOfAttractionOptions
- pydrake.systems.controllers.FiniteHorizonLinearQuadraticRegulatorOptions
- pydrake.systems.framework.UseDefaultName

Closes #13987.

I finally got fed up with commits like https://github.com/RobotLocomotion/RobotLocomotion.github.io/commit/8940aefddd1dfbe275fba1eaa9da2df5ee052367 being filled with spam, so I decided to break out some repr learning.

Usi [... cut]
PR #14993 (2021-05-03): pydrake: Add repr for RotationMatrix, RigidTransform
Closes #13987.  Relates #8586.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14993)
PR #18529 (2023-01-05): [pydrake] Add __repr__ to several classes
Closes #13987.

`RenderLabel` gets a round-trip `repr()`.

`NumericalGradientOption` lacks a getter function for one of its constructor arguments, so can only provide an incomplete `repr()`.

`SolverOptions` only provides a no-argument constructor, so cannot offer round-trip.  For the purpose of cleaning up our generated documentation, it's sufficient to just hard-code a return value in order to omit the hex address that would otherwise show up in our docs.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18529)