RobotLocomotion/drake #14011: InverseDynamicsController should not alias plant

ggould-tri, 2020-09-03:
A common pattern in our code is that we create two plants, one for simulation with oracular knowledge and complex dynamics, one with a simplified model (eg welding manipulands to end effectors) to hand to `InverseDynamicsController`.

Because `InverseDynamicsController` takes a non-owning alias of its plant, this means that its caller has to store that plant in some collateral storage with the same duration as the diagram to whose builder they passed the plant in an `AddSystem<InverseDynamicsController>` call.  This creates spaghetti ownership-flow.  This is sufficiently bad that eg we are creating a zero-sized System whose only purpose is to hold the spare plant with Diagram-scoped storage duration.

It would be better if the controller API avoided this, eg by providing an option to take ownership of the passed-in plant.

ggould-tri, 2020-09-03:
For reference, here is the workaround we used:
```
/// Trivial System to hold a single `unique_ptr` so that it will be deleted at
/// System deletion time (ie, the end of a Diagram lifespan).  Proxies the
/// get() and operator* of the underlying `unique_ptr`
template <typename T, typename Held>
class UniquePtrSystem : public drake::systems::VectorSystem<T> {
 public:
  DRAKE_NO_COPY_NO_MOVE_NO_ASSIGN(UniquePtrSystem)

  explicit UniquePtrSystem(std::unique_ptr<Held> value_to_hold)
      : drake::systems::VectorSystem<T>(0, 0),
        held_(std::move(value_to_hold)) {}

  Held* get() const { return held_.get(); }
  Held& operator*() const { return *held_; }
  Held* operator->() const { return held_.get(); }

 private:
  const std::unique_ptr<Held> held_;
};
```
```
    auto& controller_plant = *builder.AddSystem<
      UniquePtrSystem<double, MultibodyPlant<double>>>(
          std::make_unique<MultibodyPlant<double>>(0.01));
```

sherm1, 2020-09-03:
Since this requires work on InverseDynamicsController I'm thinking it's related to @huihuaTRI's issue #13718 which also proposes long-needed improvements to InverseDynamicsController. Assigning to @huihuaTRI for consideration.

huihuaTRI, 2020-09-10:
Closing this request as [PR 14037](https://reviewable.io/reviews/robotlocomotion/drake/14037#-) merged.

closed 2020-09-10 by huihuaTRI