RobotLocomotion/drake #14016: Unhelpful pydrake error when reassigning a Simulator

ggould-tri, 2020-09-03:
I ran a cell of @RussTedrake 's class notebook twice; the second time, reassigning the variable `simulator`, I got this error:

[image]

I assume this is a keepalive problem on the Simulator or something closely related.

Here's the notebook https://colab.research.google.com/github/RussTedrake/manipulation/blob/master/robot.ipynb#scrollTo=8SPr8dXoY3IT ; to reproduce run the cells up to where the text says "OK, now we're ready to simulate".  Then run the next cell twice.

jwnimmer-tri, 2020-09-03:
The context argument is passed as a `unique_ptr` (sole ownership), so passing it again is a violation of ownership.

RussTedrake, 2020-09-03:
I agree with @jwnimmer-tri that this is a fairly reasonable behavior.  I worried about presenting it that way to students... in most cases I try to keep the ownership of a system inside a single cell.  Today's notebook was the exception, because I was trying to introduce the different bits (system, context, etc) piece by piece.

BTW -- awesome that you are running the notebooks and helping like this!  ;-)

jwnimmer-tri, 2020-09-04:
Ah, I should also clarify my comment.  My statement was just a diagnosis.  As far as the direction we should point to improve this -- taking the context by `unique_ptr` in `pydrake` is fraught -- we should probably bind a keep_alive bare pointer (or `shared_ptr`) overload instead.

EricCousineau-TRI, 2020-09-05:
High-level commentary also in #13058

If we have to have 2-3 overloads for each callsite (`T*`, `unique_ptr<T>`, `shared_ptr<T>`) and possibly containers that handle those three flavors, we should either (a) make a container that make it easy to handle these cases or (b) really reconsider our adherence to `unique_ptr` in GSG.

Sure, there aren't too many bindings of accepting `unique_ptr`, but it's used in core places (e.g. `Builder.AddSystem`).

rpoyner-tri, 2020-09-08:
Assigning to @EricCousineau-TRI -- my guess from reading related tickets is that he has the most context. Feel free to bounce this @EricCousineau-TRI if I'm mistaken.

EricCousineau-TRI, 2020-09-08:
I see three issues:

1. A workflow where a context is allocated in a separate cell, and there's expectation that re-running a cell should work. That expectation is broken, and IMO you shouldn't allocate a context separate from where you pass it to a `Simulator` - that's an `underactuated` problem, so should not be relevant to Drake. (Solution: leave it as-is, allocate context in same cell, or use the Simulator's implicitly create context if you pass in `nullptr`)
2. We have `unique_ptr` usages that cause some strife. That's covered by #13058.
3. The error message for `unique_ptr` already being owned is mysterious. Adding a `repr` can help.

I will resolve this by fixing (3). Everything else should be outside the scope of this issue.

rpoyner-tri, 2026-11-11:
This looks fixed. On current master, `Simulator.__init__` in `analysis_py.cc` holds the context through a shared pointer built from the Python object (rather than taking a `unique_ptr`), so passing the same context to a second `Simulator` no longer fails with the ownership error. I checked this with the Drake 1.58.0 release wheel (Linux, Python 3.12): `Simulator(diagram, ctx)` twice with the same `ctx` raises no error. The docstring now also notes that sharing a Context among Simulators will likely give incorrect results.

Closing as completed; please reopen if you see the original error.

closed 2026-11-11 by rpoyner-tri