RobotLocomotion/drake #14026: Why python packages are installed at system level in installation scripts?

pvonmoradi, 2020-09-06:
Following the instructions [here](http://manipulation.csail.mit.edu/drake.html#section3) to install drake locally, there are 2 installation scripts which we are asked to run at `sudo` level.   
However, inspecting those scripts, we see commands for `pip install`ing some packages like ([here](https://github.com/RussTedrake/manipulation/blob/1fbec6b9995bcdc30abe28632f66109afdebf783/scripts/setup/ubuntu/18.04/install_prereqs.sh#L87)).  

From my experience, it seems that python's general workflow is against installing python packages on a user level ([installing-packages/#creating-virtual-environments](https://packaging.python.org/tutorials/installing-packages/#creating-virtual-environments)) and since in this case we are invoking that script via `sudo`, actually `sudo pip install` would be called which is discouraged [even more](https://stackoverflow.com/a/21056000/1506761).  

I'm not very experienced in application distribution and may be wrong here since it seems drake contains its own python distro, so it could be the case that somewhere in installation scripts, pip packages would be installed on drake's own PYTHONPATH.

Thanks for any clarification :D

jwnimmer-tri, 2020-09-06:
The links you post are actually for Russ's manipulation course (https://github.com/RussTedrake/manipulation), not Drake itself (https://drake.mit.edu/installation.html).

pvonmoradi, 2020-09-06:
@jwnimmer-tri Thanks for your response. 
Yes, after inspecting drake installation script itself [https://github.com/RobotLocomotion/drake/tree/master/setup/ubuntu](https://github.com/RobotLocomotion/drake/tree/master/setup/ubuntu) it seems that course's first command (`sudo /opt/drake/share/drake/setup/install_prereqs`), doesn't pip install anything (the second command does). 

Should I move this issue [there](https://github.com/RussTedrake/manipulation)?

jwnimmer-tri, 2020-09-06:
@pvonmoradi For my part, I don't mind either way.  I'll let @RussTedrake decide what to do here.

EricCousineau-TRI, 2020-09-10:
My 2 cents: It'd be best to open the issue in `manipulation` - and agreed that `sudo pip install` should be avoided!!!

Perhaps even better would be to suggest a fix in that repository, via a PR, that you can test in Colab and ensure it works from scratch (if you have time)?

(Minor relation to #1183 in terms of Colab provisioning)

RussTedrake, 2020-09-12:
I’ve pulled out the sudo pip install.  Thanks for catching; that was sloppy of me.

closed 2020-09-12 by RussTedrake