RobotLocomotion/drake #14029: drake-lcm-spy gives NoClassDefFoundError in Docker

cjds, 2020-09-06:
When I try to use `drake-lcm-spy` from within the Drake Docker I get

```
ot@cjds:/# drake-lcm-spy
ClassDiscoverer: java.lang.NoClassDefFoundError: apple/laf/AquaPopupMenuUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/aqua/AquaJidePopupMenuUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsProgressBarUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/vsnet/VsnetWindowsProgressBarUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsPopupMenuUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/windows/WindowsJidePopupMenuUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: apple/laf/AquaPopupMenuUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/aqua/AquaJidePopupMenuUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsProgressBarUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/vsnet/VsnetWindowsProgressBarUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsPopupMenuUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/windows/WindowsJidePopupMenuUI.class
Found 118 LCM types
ClassDiscoverer: java.lang.NoClassDefFoundError: apple/laf/AquaPopupMenuUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/aqua/AquaJidePopupMenuUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsProgressBarUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/vsnet/VsnetWindowsProgressBarUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsPopupMenuUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/windows/WindowsJidePopupMenuUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: apple/laf/AquaPopupMenuUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/aqua/AquaJidePopupMenuUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsProgressBarUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/vsnet/VsnetWindowsProgressBarUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsPopupMenuUI
                 jar: /opt/drake/share/java/jide-oss-2.9.7.jar
                 class: com/jidesoft/plaf/windows/WindowsJidePopupMenuUI.class
Found 0 plugins
Spy quitting
```

and a blank white screen. 
The hash of the Docker I am using is 6c5f8f3d5f28.

Reproduce (on Ubuntu 20.04 host):

```
xhost + 
docker run -it -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix robotlocomotion/drake bash
root@d7b80634dac0:/# lcm-spy
```

jamiesnape, 2020-09-08:
Best path to fixing this would be https://github.com/RobotLocomotion/drake/issues/13778, I think. As with `drake-visualizer` this is not a configuration we test, so we should be clearer about that in the documentation also.

jamiesnape, 2020-09-08:
(the `apt` packages apply a patch `debian/patches/AquaJidePopupMenuUI.java.patch` to fix this)

svenevs, 2021-11-10:
If in the docker container you run `apt-get update` so that you can `apt-get install libjchart2d-java` (which will additionally by recommendation install `libjide-oss-java` and `libxmlgraphics-commons-java`), you can edit the file `lcm-spy` to have it point to some different jars.  For example

```diff
- for jar_file in "$prefix/share/java/lcm.jar" "$prefix/share/java/jchart2d-3.3.2.jar" "$prefix/share/java/jide-oss-2.9.7.jar" "$prefix/share/java/commons-io-1.3.1.jar" "$prefix/share/java/xmlgraphics-commons-1.3.1.jar"; do
+ for jar_file in "$prefix/share/java/lcm.jar" "/usr/share/java/jchart2d.jar" "$prefix/share/java/jide-oss-2.9.7.jar" "$prefix/share/java/commons-io-1.3.1.jar" "$prefix/share/java/xmlgraphics-commons-1.3.1.jar"; do
```

(just updating `jchart2d` to the system one, leaving `jide-oss` and `xmlgraphics-commins` to the bundled versions) then the `AquaJidePopupMenuUI` messages go away.  If you update all of the jars to point to system installed ones, except for the first entry `$prefix/share/java/lcm.jar`, you'll get some of the same messages with vendored vs system `jide-oss.jar` for example

```
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsProgressBarUI
                 jar: /usr/share/java/jide-oss.jar
                 class: com/jidesoft/plaf/vsnet/VsnetWindowsProgressBarUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: sun/swing/UIClientPropertyKey
                 jar: /usr/share/java/jide-oss.jar
                 class: com/jidesoft/plaf/windows/AnimationController$PartUIClientPropertyKey.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsPopupMenuUI
                 jar: /usr/share/java/jide-oss.jar
                 class: com/jidesoft/plaf/windows/WindowsJidePopupMenuUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsSliderUI
                 jar: /usr/share/java/jide-oss.jar
                 class: com/jidesoft/plaf/windows/WindowsRangeSliderUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsProgressBarUI
                 jar: /usr/share/java/jide-oss.jar
                 class: com/jidesoft/plaf/vsnet/VsnetWindowsProgressBarUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: sun/swing/UIClientPropertyKey
                 jar: /usr/share/java/jide-oss.jar
                 class: com/jidesoft/plaf/windows/AnimationController$PartUIClientPropertyKey.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsPopupMenuUI
                 jar: /usr/share/java/jide-oss.jar
                 class: com/jidesoft/plaf/windows/WindowsJidePopupMenuUI.class
ClassDiscoverer: java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsSliderUI
                 jar: /usr/share/java/jide-oss.jar
                 class: com/jidesoft/plaf/windows/WindowsRangeSliderUI.class
```

Not sure how helpful that is, but there seems to possibly be something on the lcm side needing configuration here too? (Assuming that's what's looking for these windows classes).

jwnimmer-tri, 2022-03-14:
Most likely fixed by #16185.

closed 2022-03-14 by jwnimmer-tri