RobotLocomotion/drake #14032: Package "wsg_50_description" was found more than once in the search space.

RussTedrake, 2020-09-07:
@hjsuh94 -- I think perhaps the files you added are causing new debugging spews to the console?

```
[2020-09-07 17:14:26.349] [console] [warning] Package "wsg_50_description" was found more than once in the search space.
[2020-09-07 17:14:26.406] [console] [warning] Package "wsg_50_description" was found more than once in the search space.
[2020-09-07 17:14:26.465] [console] [warning] Package "wsg_50_description" was found more than once in the search space.
[2020-09-07 17:14:26.524] [console] [warning] Package "wsg_50_description" was found more than once in the search space.
```

https://colab.research.google.com/github/RussTedrake/manipulation/blob/master/pick.ipynb#scrollTo=oXh50Cf1hU6N

hjsuh94, 2020-09-07:
I think I know what the reason is, but the issue is probably not specific to the package `wsg_50_description`. 

There seems to be a package namespace overlap between the [models repo](https://github.com/RobotLocomotion/models) and Drake's model package. 
`@drake//manipulation/models/wsg_50_description`
`@models//wsg_50_description`

I've confirmed the same happens when loading `franka_description`.

jwnimmer-tri, 2020-09-08:
The word "package" here is not bazel packages like ` @models//wsg_50_description`.  It means ROS packages ala `package.xml` files.  I don't think the models repository has a `package.xml` file.  Something else must be going on.

hjsuh94, 2020-09-08:
As always, thanks for the input @jwnimmer-tri - I tried messing with bazel settings but haven't been successful, so I was kind of lost on what the issue was. 

As you pointed out, it turned out that some models like `franka_description` and `wsg_50_description` has `package.xml` files in them in Drake, loading these `sdf` files from these folders cause errors. Other models (`ycb`, `jaco_description`) don't seem to have this problem because they don't have `package.xml` in them. 

I'm not sure where this inconsistency across models comes from, or what a good fix should be - is it an artifact of using `xacro`?

jwnimmer-tri, 2020-09-08:
To debug this, I'd suggest improving the error message to elaborate on what the "duplicate" paths were.

hjsuh94, 2020-09-08:
Some updates on debugging - I've confirmed there are no duplicate paths and they are always pointing to the right path, but simply trying to add the same model twice through `Parser` causes the error if that model has a `package.xml` file inside. 

I've managed to reproduce the error in a very minimal setting in this notebook:
https://colab.research.google.com/drive/1oUrHlB4HJUngbM_b5Kdoz9BDkqzrw6or?usp=sharing

The "backtrace" of the error happens as follows:

1. When [`AddModelFromFile()`](https://github.com/RobotLocomotion/drake/blob/d7be0e0ba3dee5675dba8494b16738b58da793c7/multibody/parsing/parser.cc#L52) is called from Parser, it calls [`PopulateUpstreamToDrake`](https://github.com/RobotLocomotion/drake/blob/d7be0e0ba3dee5675dba8494b16738b58da793c7/multibody/parsing/package_map.cc#L157) from the package map. 
2. `PopulateUpstreamToDrake` calls [`PopulateUpstreamToDrakeHelper`](https://github.com/RobotLocomotion/drake/blob/d7be0e0ba3dee5675dba8494b16738b58da793c7/multibody/parsing/package_map.cc#L136), which calls [`AddPackageIfNew`](https://github.com/RobotLocomotion/drake/blob/d7be0e0ba3dee5675dba8494b16738b58da793c7/multibody/parsing/package_map.cc#L123)
3. `AddPackageIfNew` complains because this package was already added into the parser. 

I suppose it's a reasonable thing to 'warn' about?

jwnimmer-tri, 2020-09-10:
If the user _directly_ makes package-path-related API calls that are silly, then yes we could warn them about it.  However, merely asking adding the same SDFormat model to a plant twice in a row should not warn.

(I think a relevant factor here is re-using the Parser for both additions.  Making a new Parser every time would be a work-around.)

It sounds like (1) we should fix the Parser logic to avoid emitting this warning and (2) this probably was not some new regression related to the WSG 50 mesh changes, but rather adding something twice with the same parser object.

Per (2), you are welcome to opt-out of this ticket if you'd rather not work on fixing it.

hjsuh94, 2020-09-10:
Unless the fix is to delete the line that warns, I would prefer not going through parser logic.

PR #14089 (2020-09-21): parsing: Don't warn when re-discovering the same package.xml
As best I can tell, this bug has been around ~forever.

Closes #14032.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14089)
closed 2020-09-21 by jwnimmer-tri