RobotLocomotion/drake #14033: multibody: `ModelInstanceIndex` as an optional argument to the Jacobian methods.

RussTedrake, 2020-09-08:
Working with Jacobians in a MBP that has multiple instances is a pain -- it's up the user to pull out the elements of the jacobian that they actually need.  ([here is an example](https://colab.research.google.com/github/RussTedrake/manipulation/blob/master/pick.ipynb#scrollTo=ue9ofS7GHpXr))

It seems quite reasonable to support an optional argument for `ModelInstanceIndex` that would just compute/return the gradients wrt a subset of the qdot or v for a particular model.  This could have even more value/meaning for the `CalcJacobianCenterOfMassTranslationalVelocity` method (if it would compute CM for just that model instance).

xuchenhan-tri, 2022-05-24:
@RussTedrake Is the request to provide sugar that extracts relevant parts of the Jacobian or to provide a more efficient implementation (aka avoid irrelevant calculations)? My understanding is the former, but a confirmation would be helpful.

If so, a fix similar to that in #15900 should be easy and I can take on this issue in the coming weeks. 

BTW, `CalcJacobianCenterOfMassTranslationalVelocity` for selected model instances is already supported via #15387.

xuchenhan-tri, 2022-05-24:
Per f2f, @mitiguy will pick up this ticket upon confirmation from OP that this is still relevant.

RussTedrake, 2022-05-25:
The request was to do the same as #15387 for any remaining jacobians in multibody plant.  Of course saving calculations is great, but this was a usability issue first not a performance issue.  If computing and then filtering is the best first approach, that could be ok.

xuchenhan-tri, 2022-05-25:
Thanks for the confirmation. The usability issue is an easy fix as shown in #15900. I put it on the MbP hitlist.

This is slightly different from #15387 though. There we were cutting down the number of functions being differentiated whereas here the request is to cut down the w.r.t. variables.