RobotLocomotion/drake #14044: Relocate example binaries, data, and related licenses in binary packages

jamiesnape, 2020-09-10:
Basically everything goes into `/opt/drake/share/drake/examples/<example_directory>` at present. A preferable situation (since `share` is conventionally for platform independent files) would be binaries in `/opt/drake/lib[/x86_64-linux-gnu]/drake/examples/<example_directory>` for binaries, `/opt/drake/share/drake/examples/<example_directory>` for data (unchanged), and `/opt/drake/share/doc/drake/examples/<example_directory>` for licenses.

Relates #12783.

jamiesnape, 2020-09-10:
Note in Debian packaging terms, examples typically refers to source code examples, so our compiled examples do trigger warnings, but those are trivial to squash. Also, we have specific reasons for not throwing the binaries into `/opt/drake/bin` (mostly because we may have filename conflicts without the directory tree).

jwnimmer-tri, 2020-09-16:
In theory, I would love to follow the FHS and keep "share" as platform-independent only, with the binaries in "lib".  However, our use of data resources (and resource finding) does not distinguish whether the resource being sought is platform independent or not.  If I want to call `drake::FindResource` on `common/resource_tool`, that still needs to work.  Shuffling around the installation paths to placate the Debian linter does not seem to be very valuable, given the development effort and user distress that it will cause.  Is there any practical downside to diverging from the standard?  We are already in `/opt/drake` all by our lonesome, it doesn't seem like it should matter what paths we choose?

jamiesnape, 2020-09-16:
`resource_tool` is not under `examples`, surely? No linter placating, we already suppress several thousand complaints and would make the same comment for the Mac packages. All our other licenses are under `share/doc`, so seems logical enough the others should be there, but if it is too much churn, it is really not that important.

jwnimmer-tri, 2020-09-16:
I have no objection to reorganizing where the license texts end up.

I think my statement still applies to moving the example binaries, though.  The example binaries can be sensible `FindResource` targets.  Is there any tangible benefit to changing the installation paths of our programs?  Is there any practical downside to leaving things alone (and so diverging from the "platform independent `share`" standard)?

jamiesnape, 2020-09-16:
Just to summarize Slack here: data and binaries need to stay in the same tree for resource finding to work; moving everything to `lib/<arch>` may be helpful if/when we support multiple architectures; data in `lib/<arch>` may be as confusing as binaries in `share`, so probably not a big win there.

Save for the license fix up, assuming it is easy, probably should send this issue to the backlog or close it.

jamiesnape, 2020-09-16:
https://github.com/RobotLocomotion/drake/issues/4224#issuecomment-693144801 is probably the future, so let's just close this for now.

jwnimmer-tri, 2021-11-07:
> ... let's just close this for now.

closed 2021-11-07 by jwnimmer-tri