RobotLocomotion/drake #14047: pybind: Should restore fork update pending mac fix

EricCousineau-TRI, 2020-09-11:
See: https://github.com/RobotLocomotion/drake/pull/14046#issuecomment-690834436

@jamiesnape I will try this briefly. If I can't make this work today, can I ask you to put this at the top of your `pydrake` queue for the next two weeks?

jamiesnape, 2020-09-11:
So this issue is restoring the fork or fixing the underlying Mac issue?

EricCousineau-TRI, 2020-09-11:
Fixing the underlying Mac issue for building in CMake that causes the RTTI mismatch, most likely due to some visibility changes that came from the fork updates.
Not sure if it requires a change to our fork, or just a change in downstream.

(RobotLocomotion/pybind11 itself is unchanged at present, and I don't think we should revert anything there)

EricCousineau-TRI, 2020-09-11:
FTR On `macsim`, I am using this `drake-external-examples` setup:
https://github.com/RobotLocomotion/drake-external-examples/blob/8a53b4c8df534fcc02f557d1579636dd16795151/drake_cmake_installed/src/simple_bindings/CMakeLists.txt

I confirmed that drake @ 690a7eacd1 works (post-revert), and drake @ 0cc01f0c7 reprodues the error:

```sh
cd .../drake
bazel run //:install -- /tmp/drake

cd .../bazel-external-examples/drake_cmake_installed
mkdir build && cd build
cmake -DCMAKE_PREFIX_PATH=/tmp/drake ..
cd src/simple_bindings/
make && ctest -V -R
```

jamiesnape, 2020-09-11:
Ok, can you update the title to match?

EricCousineau-TRI, 2020-09-11:
Done.

My money's on some change from upstream's CMake in `pybind11Tools` or `pybind11Config`:
https://github.com/RobotLocomotion/pybind11/commit/58a368ea8#diff-8cb32abb960df57289d52b400092e31d

EricCousineau-TRI, 2020-09-11:
Yuppers. It looks like https://github.com/pybind/pybind11/commit/97aa54fefa is the culprit.
I had stripped it out (accidentally?) before, and restored it when I carte-blanch took in upstream's changes.

Here's the patch-fix for `drake-external-examples`:
```patch
diff --git a/drake_cmake_installed/src/simple_bindings/CMakeLists.txt b/drake_cmake_installed/src/simple_bindings/CMakeLists.txt
index 1fde04b..5aa2aa4 100644
--- a/drake_cmake_installed/src/simple_bindings/CMakeLists.txt
+++ b/drake_cmake_installed/src/simple_bindings/CMakeLists.txt
@@ -33,7 +33,12 @@
 find_package(pybind11 CONFIG REQUIRED)
 
 pybind11_add_module(simple_bindings MODULE simple_bindings.cc)
-target_link_libraries(simple_bindings PUBLIC drake::drake pybind11::module)
+target_link_libraries(simple_bindings PUBLIC drake::drake)
+# N.B. `pybind11_add_module` normally sets the default visibility to "hidden"
+# to avoid warnings. However, we generally take care in Drake to avoid mixing
+# hidden and public symbols, so we should we confidently re-enable this
+# setting.
+set_target_properties(simple_bindings PROPERTIES CXX_VISIBILITY_PRESET "default")
 
 add_test(NAME simple_bindings_test COMMAND
     "${PYTHON_EXECUTABLE}" "${CMAKE_CURRENT_SOURCE_DIR}/simple_bindings_test.py")
```

My desire is to leave upstream's CMake logic as untouched as possible. I think just propogating this specific `CXX_VISIBILITY_PRESET` post-fix should be fine.

EricCousineau-TRI, 2020-09-11:
Confirmed that this patch works on both drake @ 690a7ea and drake @ 0cc01f0. Will update overview with plan of attack.

EricCousineau-TRI, 2020-09-11:
Filed upstream issue to pybind11: https://github.com/pybind/pybind11/issues/2479

PR #14048 (2020-09-11): Restore "workspace: Update pybind11 fork to latest commit"
**Downstream Warning**:
If you are using CMake to consume `pydrake` and you compile your own bindings using `pybind11_add_module`, you must now reset your target's default visibility to "default" (not "hidden").
See [drake_cmake_installed](https://github.com/RobotLocomotion/drake-external-examples/blob/9a70157/drake_cmake_installed/src/simple_bindings/CMakeLists.txt#L45) for an example of doing so.

This reverts commit 690a7eacd136c842d2a7aeda7a5228cfa9d7af95.

Resolves #14047

---
This change is [[image]](https://reviewable.io/reviews [... cut]
closed 2020-09-11 by EricCousineau-TRI