RobotLocomotion/drake #14051: fishy-looking math in AutoDiffScalar atan()

rpoyner-tri, 2020-09-11:
Questions were raised in the context of #14045 about the details of the math implementing atan() for AutoDiffXd. I couldn't answer without a bit of history, and some way to evaluate the alternatives. This issue is to track that follow-up.

Comments:
https://github.com/RobotLocomotion/drake/pull/14045#pullrequestreview-487016357
and 
https://github.com/RobotLocomotion/drake/pull/14045#pullrequestreview-487052140

sherm1, 2020-09-11:
Assigned to @mitiguy for a look. Can you tell if the Eigen atan/asin/etc derivatives are correct for all scalar types? The implementations seem inconsistent.

jwnimmer-tri, 2020-09-11:
The file in that review is only used with `Scalar = double`.  Was the code suspicious in even with doubles?

sherm1, 2020-09-11:
> The file in that review is only used with Scalar = double. Was the code suspicious when only used with doubles?

No, I think it is just inconsistently implemented in that case.

jwnimmer-tri, 2021-11-11:
So @rpoyner-tri how about you change `atan` to use `abs2` for consistency, and then call it a day?  Or else decide we don't care and close the issue.  Either way, this is not worth lingering around.

sherm1, 2021-11-11:
Per discussion in #16075 I believe the implementation with abs2() is incorrect, though the error would only be apparent for complex numbers. Other uses of abs2() there also seem wrong to me. If we want consistency, we should fix the ones that use abs2() to use plain old square instead of absolute square.

PR #16075 (2021-11-11): autodiffxd atan(): Use abs2()
Closes #14051.

This is mostly to answer complaints about inconsistencies with similar
functions.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16075)
commit 92bdeee9 (2021-11-11): autodiffxd atan(): Use abs2() (#16075)

* autodiffxd atan(): Make consistent with other implementations

Closes #14051.

This is mostly to answer complaints about inconsistencies with similar
functions. Also, it answers complaints about confusion over the use of
abs2().
closed 2021-11-11 by sherm1