RobotLocomotion/drake #14052: fishy math in AutoDiffScalar pow()

rpoyner-tri, 2020-09-11:
Noticed in this PR: #14045 

pow()'s math implementation is probably not as fast as it could be. This ticket is to track the follow-up on that observation.

See comment thread here: https://reviewable.io/reviews/robotlocomotion/drake/14045#-MGyJSDR1w6N05T5ODEE:-MGyvlrPAmEoRtQ_rqec:b2pgy9o

sherm1, 2020-09-11:
Here the issue is just performance (not yet of proven significance though). We're not saying this produces the wrong answer -- it just invokes pow() twice when it could do it once.

jwnimmer-tri, 2021-11-11:
There is already a TODO in the code about this (citing this issue number).  I don't know that having an open issue in our backlog is particularly useful?  The mean number of bugs per line of code is approximately 1.0, and so tracking them all with issues is going to be overwhelming.  Is there any particular reason to think that _this_ particular subject matter requires our ongoing and dutiful attention in monthly bug scrubs?  If not, maybe we can close and save us some trouble.

jwnimmer-tri, 2022-07-29:
There were two concrete proposals in that PR:

(1) From Alejandro:
```
a.value() = pow(a.value(), b);
a.derivatives() *= b / a * a.value();
```

(2) From Sherm:
```
const Scalar ab1 = pow(a.value(), b-1);
a.value() = a * ab1;        // a^b
a.derivatives() *= b * ab1; // b a^(b-1)
```

The key point here will be the IEEE special cases documented in https://en.cppreference.com/w/cpp/numeric/math/pow.

---

Consider proposal (2) first -- it's the easier of the two to rule out.

Is the proposed `result.value() = base * pow(base, exp-1)` equivalent to the true value `result.value() = pow(base, exp)`?

With real numbers it is, but in IEEE floating point it's not.  For example, with `base == +0` and `exp == +0` the answer should be `pow(0, 0) ==> 1` but with (2) we get `0 * pow(0, -1) ==> 0 * inf ==> NaN`.

---

Consider proposal (1).

Is the proposed `result.derivatives() *= exp * pow(base, exp) / base;` equivalent to `result.derivatives() *= exp * pow(base, exp-1);`?

For example with `base == +inf` and `exp == 0` the scaling should be `0 * pow(inf, 0-1) ==> 0 * 0 ==> 0` but with (1) we get `0 * pow(inf, 0) / inf ==> 0 * 1 / inf ==> ...` either `0` or `NaN` depending on the order of operations.

I think _maybe_ `a.derivatives() *= b * (new_value / a);` is correct but seems too arcane to be worth trying to use an optimization.  It's also not clear to me whether the cost of `pow+div+mul` in serial is cheaper than `pow | pow+mul` in parallel.

---

In any case, all this goes out the window when `exp` has gradients.  My plan is to write a proper and well-tested version `pow()` in my next AutoDiff PR.

jwnimmer-tri, 2022-07-30:
My proposal is that #17655 should be sufficient to close this.  The correct implementation is pushing 200 lines long.  I don't think we should backport that to `AutoDiffXd`.

PR #17655 (2022-08-18): [ad] Add support for power functions
Towards #17492.  Closes #14052.

---

FYI for your reference on `pow()` special cases, a screenshot from my [Google Sheet](https://docs.google.com/spreadsheets/d/1HWKwWr2HySZU_vohyJ2o0qMLbGcZEEOPznOsK5x_iGw/edit#gid=0) with the table of result.  The sheet also shows grad(base) and grad(exp) validity in the secondary tabs.

[image]

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17655)
closed 2022-08-18 by jwnimmer-tri