RobotLocomotion/drake #14054: Broken links in MultibodyPlant documentation

jamiesnape, 2020-09-13:
Note the local build path `/media/ephemeral0/ubuntu/workspace/linux-bionic-gcc-bazel-nightly-documentation` of the documentation job in the links on this line:

https://github.com/RobotLocomotion/RobotLocomotion.github.io/blob/9856f98a652ad72d4aa9649c144d846d48911d7d/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#L419

Generated from https://github.com/RobotLocomotion/drake/blob/6bd5f8c25cdcbb4049e57ec0f49f6e2c136d864a/multibody/plant/multibody_plant.h#L1411-L1412.

Rendered as https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html. Search for line "define them in an input URDF/SDF as detailed here for SDF or here for URDF".

jamiesnape, 2020-09-13:
Note that this triggers the following lintian warning when packaging drake:

`I: drake-doc: file-references-package-build-path opt/drake/share/doc/libdrake-doc/html/classdrake_1_1multibody_1_1_multibody_plant.html`

sherm1, 2020-09-13:
That `@ref urdf_contact_material` in `multibody/plant/multibody_plant.h` is intended to reference `@anchor urdf_contact_material` in `multibody/parsing/detail_urdf_geometry.h:137`. Any idea why that would generate a local URL?

Right in the same spot is `@ref sdf_contact_material` referencing `@anchor sdf_contact_material` in `multibody/parsing/detail_scene_graph.h:124`. Does that have the same problem?

jamiesnape, 2020-09-14:
> That @ref urdf_contact_material in multibody/plant/multibody_plant.h is intended to reference @anchor urdf_contact_material in multibody/parsing/detail_urdf_geometry.h:137. Any idea why that would generate a local URL?

Not yet. We do have many examples of `@ref` working correctly, so need to work out what is different here.

> Right in the same spot is @ref sdf_contact_material referencing @anchor sdf_contact_material in multibody/parsing/detail_scene_graph.h:124. Does that have the same problem?

Yes.

EricCousineau-TRI, 2020-09-22:
#14107 is a duplicate of this issue, but @SeanCurtis-TRI I think has identified the root cause in that one. Will close this for now.

closed 2020-09-22 by EricCousineau-TRI