RobotLocomotion/drake #14055: Support open-source Linear Programming solver

hongkai-dai, 2020-09-13:
Currently we don't have a very good open-source LP solver in drake.

Some candidates are

- CLP https://github.com/coin-or/Clp, this uses EPL-2. I don't know if we support EPL-2. In https://github.com/RobotLocomotion/drake/issues/4045 we support EPL-1. 
- GLOP from google https://github.com/google/or-tools/tree/stable/ortools/glop. It uses Apache License.

@jamiesnape and @jeremyma-tri , do we support EPL-2 license?

jamiesnape, 2020-09-14:
> @jamiesnape and @jeremyma-tri , do we support EPL-2 license?

Yes.

jamiesnape, 2020-09-14:
And we already use `Clp` as a transitive dependency.

hongkai-dai, 2021-03-03:
I would like to add CLP into Drake. @jamiesnape , could you point me to how I should get started to add its build file into drake? Is there a template inside Drake I could use? Thanks a lot!

jwnimmer-tri, 2021-03-03:
It's in Ubuntu as https://packages.ubuntu.com/bionic/coinor-libclp-dev and we already depend on it (per dReal, noted upthread):
https://github.com/RobotLocomotion/drake/blob/faa2301c3715781b4b9cee0a70b62302a4fb95f3/setup/ubuntu/source_distribution/packages-bionic.txt#L3

To bring it into workspace, it will look similar to how we do `tools/workspace/ipopt`.  On a quick glance, it seems like the only changes would be `modname = "clp"` and the `licenses` block.

hongkai-dai, 2021-03-03:
Thanks @jwnimmer-tri ! I just sent PR https://github.com/coin-or/Ipopt/blob/master/LICENSE. I am not sure if I need to change the licenses block, as both CLP and IPOPT use the same EPL 2.0 license https://github.com/coin-or/Ipopt/blob/master/LICENSE?

jwnimmer-tri, 2021-03-03:
FYI I'll switch my suggestions over to the PR code, which will be an easier place to discuss.

hongkai-dai, 2021-04-16:
We added CLP to Drake.

closed 2021-04-16 by hongkai-dai