RobotLocomotion/drake #14056: Cannot (yet) use RenderEngineVTK on Colab

RussTedrake, 2020-09-13:
My simple test notebook
https://colab.research.google.com/github/RussTedrake/manipulation/blob/master/pose.ipynb
is crashing... seemingly because it is trying to open a window with a bad X server connection? 

[image]

Note that I *am* passing `show_window=False` into the `RgbdSensor` constructor, and I don't see any windows open when I run this locally.  I believe we've addressed these issues for running headless before.  Is there a magic incantation that I am missing?

jwnimmer-tri, 2020-09-13:
For reference, both Drake CI and Anzu CI provide a `DISPLAY` variable that works.

Anzu CI uses a real xorg server with GPU hardware.

Drake CI does this:
- https://github.com/RobotLocomotion/drake-ci/blob/f34fb0054aa50edc40fbae3a907c85651d6da8ce/setup/ubuntu/install_prereqs#L78-L87
- https://github.com/RobotLocomotion/drake-ci/blob/f34fb0054aa50edc40fbae3a907c85651d6da8ce/driver/platform.cmake#L59-L61

Some tests require DISPLAY, even though nothing opens a window.

I'm not sure if there is a way to use the renderer without X.  Maybe with mesa something.  You should be able to test locally by unsetting DISPLAY and seeing the same result.

RussTedrake, 2020-09-13:
This looks promising.  https://colab.research.google.com/github/tensorflow/lucid/blob/master/notebooks/differentiable-parameterizations/appendix/colab_gl.ipynb

RussTedrake, 2020-09-14:
It looks like I should be able to use xvfb.  https://colab.research.google.com/drive/1flu31ulJlgiRL1dnN2ir8wGh9p7Zij2t .  Am giving that a spin now.

RussTedrake, 2020-09-14:
Woot!  That's solved it.  Thanks @jwnimmer-tri !

closed 2020-09-14 by RussTedrake

mudamuda-can, 2021-07-12:
https://www.programmersought.com/article/42158925287/