RobotLocomotion/drake #14058: drake::math::expmap2quat() should use std::pow().

guzhaoyuan, 2020-09-13:
After upgrading to Drake v0.16.0, my drake::math::expmap2quat() function call fails in compilation stage. Under this drake::math::expmap2quat() function, the [pow() function](https://github.com/RobotLocomotion/drake/blob/6bd5f8c25cdcbb4049e57ec0f49f6e2c136d864a/math/expmap.h#L68) should be std::pow(). 
After some look up, I think the cause is: the implementation of another pow() function (https://github.com/RobotLocomotion/drake/pull/12835) confuses the compiler.

jwnimmer-tri, 2020-09-14:
For a fix, within the `expmap2quat` function body the code should say `using std::pow;`.

@hongkai-dai No code within Drake itself uses `expmap.h` anymore.  Do you have a sense of how useful it is to keep around?  If nobody is maintaining it, and it's not useful for the future, perhaps we should consider deprecating and then removing it.

hongkai-dai, 2020-09-14:
I think we should zap it. I wrote that code from 5 years ago, and never visited it again.

jwnimmer-tri, 2020-09-14:
@guzhaoyuan FYI If the functions are still useful for your work, then Drake's license permits you to copy the sources into your own project to keep them around and/or fix them.

guzhaoyuan, 2020-09-14:
Thanks for letting me know!

EricCousineau-TRI, 2020-09-14:
Per checklist (and our issue management policies), all issues must have owners:
https://drake.mit.edu/platform_reviewer_checklist.html

Assigning @hongkai-dai for now; please delegate if you need help!

PR #14099 (2020-09-21): math: Deprecate expmap.h
Closes #14058.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14099)
closed 2020-09-21 by SeanCurtis-TRI