RobotLocomotion/drake #14068: Issue with drake_werror and drake_cc_vector_gen_library

cjds, 2020-09-15:
Trying to use `drake_cc_vector_gen_library`  in an external Drake project and I run into the following error

```
ERROR: Analysis of target '//double_pendulum:energy_shaping' failed; build aborted: no such package 'tools': BUILD file not found in any of the following directories. Add a BUILD file to a directory to mark it as a package.
```

When I create the folder tools and the BUILD file

```
ERROR: While resolving configuration keys for //double_pendulum:pendulum_vector_types: no such target '//tools:drake_werror': target 'drake_werror' not declared in package 'tools' defined by /root/doublependulum/tools/BUILD
ERROR: Analysis of target '//double_pendulum:energy_shaping' failed; build aborted: no such target '//tools:drake_werror': target 'drake_werror' not declared in package 'tools' defined by /root/doublependulum/tools/BUILD
```
Probably coming from 

https://github.com/RobotLocomotion/drake/blob/7c06b9127077e10aa8444ba61aacae687408d3b6/tools/skylark/drake_cc.bzl#L66

I think

Repro using

```
drake_cc_vector_gen_library(
    name = "pendulum_vector_types",
    srcs = [
      "pendulum_state_named_vector.yaml",
    ],
    visibility = ["//visibility:public"],
)

```

and `pendulum_state_named_vector.yaml`
```
namespace: drake::examples::pendulum
elements:
  -
    name: theta
    doc: The angle of the pendulum.
    doc_units: radians
    default_value: 0.0
  -
    name: thetadot
    doc: The angular velocity of the pendulum.
    doc_units: radians/sec
    default_value: 0.0

```

jwnimmer-tri, 2020-09-15:
My (probably completely undocumented) intent for bazel macros is that `drake_foo_bar` macros are reserved solely for use within Drake, and just plain `foo_bar` is what downstream projects should use.  However, in the case of `vector_gen` we don't offer a sufficient variety of macros to make the alternatives possible.  I'll work on adding those.

cjds, 2020-09-17:
So looking at it , the thing that I'm struggling with is because these macros are used in the examples and because I'm trying to copy the example as an external bazel repository it makes it hard for me to run them.

Would you recommend me making a copy of the `gen` rule in my repository or is there a way to view the C++ that it produces so I can run it in `drake` and copy into mine

jwnimmer-tri, 2020-09-17:
Yes, if you copy https://github.com/RobotLocomotion/drake/blob/master/tools/vector_gen/vector_gen.bzl and change its call of `drake_cc_library` into just `cc_library`, that's all you'll need.

closed 2020-09-21 by jwnimmer-tri

cjds, 2020-09-27:
@jwnimmer-tri is there a good reason not to make it something usable outside of drake?

The reason I'm asking is it is used in quite a few of the examples eg.

https://github.com/RobotLocomotion/drake/blob/master/examples/pendulum/pendulum_params_named_vector.yaml#L1

among other places

but that means I can't easily port any of these examples into the external project examples

https://github.com/RobotLocomotion/drake-external-examples

---

Ideally I'd like to be able to run the examples in an external environment without copying the tool into that env

jwnimmer-tri, 2020-09-27:
As I understand it, there is no meaningful impediment to using named vector yamls in downstream projects.  I believe the `vector_gen` binary and the `cc_vector_gen` macro are directly usable by downstream projects, so there is no need to copy the tool.

The only thing downstream projects might need to re-implement is the `drake_cc_vector_gen_library` macro.  Almost every line in that macro would change for downstream use, and so the Drake copy of the macro is not useful for other projects to call.  It does not seem like an undue burden to rewrite those few lines?

Or maybe I'm misunderstanding the problem, and you could clarify?

cjds, 2020-10-05:
I see I think I don't quite understand why `Almost every line in that macro would change for downstream use` but I suspect as I use it I might understand more

PR #14092 (2020-09-21): vector_gen: Clarify which macros are to be used outside of Drake
This renames some macros (providing deprecation stubs in the meantime).

Closes #14068.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14092)