RobotLocomotion/drake #14077: doc/system_doxygen.py assumes locale supports UTF-8

jamiesnape, 2020-09-16:
We do ensure that `en_US.UTF-8` is available, but for Doxygen we do not actually enforce that the selected locale is `en_US.UTF-8`. Most people would not notice (people using Docker or a minimal chroot would), but we do cover the eventuality when using Sphinx, so would be best to do so here as well.

(Aside: Really we only need `something.UTF-8` and `C.UTF-8` may even be good enough. Not a great hardship to install one extra locale, though.)

RussTedrake, 2020-11-13:
i think we can change system_doxygen.py to whatever it needs to conform.  there was no intent to assume any extra locale support here.

jwnimmer-tri, 2021-11-10:
I think the call to action here is to move this one line:

https://github.com/RobotLocomotion/drake/blob/1fe51826673ee24feea1b80baf1b950cc577bfe5/doc/pydrake/build.py#L155

... down into the `drake/doc/defs.py` helper that wraps subprocess.  Should be easy.

jwnimmer-tri, 2022-01-27:
Big picture: we want our Documentation builds to be robust (reproducible) wrt the user's environment.  To the extent that the locale settings (LANG, LC_ALL, whatever) impact any documentation output, we should hard-code them to be `en_US.UTF-8` for any and all sub-processes that we launch.  I _think_ there is a single function in `build.py` that launches all docs subprocesses, so setting the environment there should be sufficient to call this complete.  We don't necessarily need to test the fix in a Docker with a bad-locale, though doing so is fine if it's easy.

mwoehlke-kitware, 2022-01-28:
What is actually going wrong, here? How can this be reproduced?

I generated dox on a Docker image that appears to have no UTF8 locales installed, and with `LC_ALL=C`, and UTF-8 characters still seem to be rendered okay.

jwnimmer-tri, 2022-01-28:
There is no reproduction recipe required.  Change the code here [1] to set  `env["LC_ALL"] = "en_US.UTF-8"` when launching the subprocess (in both branches), so that our documentation builds are reproducible.  Check that the generated output has not regressed, by comparing the new output with either a build without that change, or the content of the deployed site (https://github.com/RobotLocomotion/RobotLocomotion.github.io).

[1]
https://github.com/RobotLocomotion/drake/blob/a172633cd35fa1e3fa68d3dac73af405c91857c2/doc/defs.py#L70-L75

PR #16474 (2022-02-01): Ensure Doxygen runs in a UTF-8 locale
Fixes #14077.

+@jwnimmer-tri for feature review, please.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16474)
closed 2022-02-01 by jwnimmer-tri