RobotLocomotion/drake #14085: Xcode 12 support

jamiesnape, 2020-09-17:
[Xcode 12 is out now](https://developer.apple.com/news/?id=qbomldbb), and will be the minimum supported version for macOS Big Sur 11.0 as well as the version most macOS Catalina 10.15 users have due to Apple's aggressive update settings. Apple Clang 12 appears to descent from mainline Clang 10, which is newer than our present supported version 9 on Ubuntu. This means, mostly, more warnings (and therefore warnings as errors).

Now, our code generally compiles with Xcode 12, since early work porting to Ubuntu 20.04 Focal used the default Clang 10, but many of the issues fixed them have crept back into the codebase since. In particular, from https://github.com/RobotLocomotion/drake/issues/13648#issuecomment-654911710:

> We would need to decide what to do about -Wrange-loop-analysis warnings/errors. They are trivial to fix, but not something that GCC, Clang 9, or Xcode 11 picks up.

We currently do not have a particularly good way to switch off that particular warning since it is not a flag that existed prior to Xcode 12 and mainline Clang 10, and our currently supported (for a month or two) macOS High Sierra 10.15.14 will never support Xcode 12, so on the face of it fixing `-Wrange-loop-analysis` and other warnings would be the easiest route, whether or not people find the warnings objectionable (https://github.com/RobotLocomotion/drake/pull/13357). However, most of our Ubuntu users would not trip the warnings until their code merged into master and failed on nightly CI, which is not good. Fortunately, there is now [Clang 10 for Ubuntu 18.04 Bionic](https://packages.ubuntu.com/bionic/clang-10), but another compiler change in the last six months is not ideal, and ultimately (but not necessarily coupled with this) we would want a matching `clang-format` version which is further churn.

Relates #13648, but there may also be standard C++ library changes in the new platform SDK that make it a different issue to this.

FYI @jwnimmer-tri

jwnimmer-tri, 2020-09-17:
It doesn't seem too difficult to expand `load("@cc//:compiler.bzl", "COMPILER_ID")` as used by `drake_cc.bzl` to also propagate the major + minor versions that `identify_compiler` already computes to be part of the bzl constants, so that we can have vendor- and version-specific `CLANG_FLAGS`.  That's what I would try first -- hopefully a 30 line PR that unblocks other work.

After that, though, I think my tendency is actually to switch Ubuntu to Clang 10 at some point after current work (at which point we could re-enable this warning).

jamiesnape, 2020-09-17:
True, forgot we had a version check.

closed 2020-10-01 by jamiesnape