RobotLocomotion/drake #14107: hydroelastic: Links to SDFormat and URDf specification are broken?

EricCousineau-TRI, 2020-09-22:
As of master @ 598ab9c303, I tried to click on the following links:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#mbp_hydroelastic_materials_properties
> [image]

But those links appear to be broken:
```
define them in an input URDF/SDF as detailed <a class="el" href="/media/ephemeral0/ubuntu/workspace/linux-bionic-gcc-bazel-nightly-documentation/src/doxygen_tmp/input/drake/multibody/parsing/detail_scene_graph.h.html#sdf_contact_material">here for SDF</a> or <a class="el" href="/media/ephemeral0/ubuntu/workspace/linux-bionic-gcc-bazel-nightly-documentation/src/doxygen_tmp/input/drake/multibody/parsing/detail_urdf_geometry.h.html#urdf_contact_material">here for URDF</a>.
```

EricCousineau-TRI, 2020-09-22:
Here's that text in source:
https://github.com/RobotLocomotion/drake/blob/598ab9c303b9fb1782b86295c03af882fd098e4f/multibody/plant/multibody_plant.h#L1412-L1413

And here're the two anchors that are referred to:
https://github.com/RobotLocomotion/drake/blob/598ab9c303b9fb1782b86295c03af882fd098e4f/multibody/parsing/detail_urdf_geometry.h#L137
https://github.com/RobotLocomotion/drake/blob/598ab9c303b9fb1782b86295c03af882fd098e4f/multibody/parsing/detail_scene_graph.h#L124

FYI @SeanCurtis-TRI

SeanCurtis-TRI, 2020-09-22:
I'm surprised they render as links. They end up linking into internal code. In my mind, this is part of #13314 (although it's not explicitly enumerated there).

jwnimmer-tri, 2020-09-22:
Is this a duplicate of #14054?

EricCousineau-TRI, 2020-09-22:
Ah, makes sense. Reassigning, as it's just a "What's documented publicly" question.

EricCousineau-TRI, 2020-09-22:
@jwnimmer-tri Yup, seems so. Though this issue, with Sean's reference to #13314, I'd say the root-cause is a bit more clear?

SeanCurtis-TRI, 2021-12-04:
This should get resolved by writing the user guide specified in #15796. I'm leaving it open until then.

rpoyner-tri, 2022-02-01:
Another possibility is to link both of them to the drake:proximity_properties section in the parsing reference. Such a resolution would depend on #16444.

PR #16522 (2022-02-05): [doc] Fix broken links in multibody doc
Closes: #14107

Revise links to point to the parsing reference, and modernize
instances of "SDF" to "SDFormat".

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16522)
commit d302fe25 (2022-02-05): [doc] Fix broken links in multibody doc (#16522)

Closes: #14107

Revise links to point to the parsing reference, and modernize
instances of "SDF" to "SDFormat".
closed 2022-02-05 by jwnimmer-tri