RobotLocomotion/drake #14114: linear_bushing_roll_pitch_yaw: Users may encounter gimbal lock, in odd ways?

EricCousineau-TRI, 2020-09-23:
@sherm1 noted this in an email, but this is prolly obvious. The only non-obvious thing to me was that it took me the following reproduction recipe to easily capture a gimbal lock error message:
<Will update>

I was able to reproduce this on Anzu's latest master (as of time of writing), as well as drake @ ec5de34e0b75ff8e9ed80d0207020c11e26ec7ae (the latest nightly binary).

Dunno if there's a good workaround other than to either avoid it (easy) or use a different internal representation for the rotational DoFs (harder-ish).

EricCousineau-TRI, 2020-09-23:
FWIW This can be encountered if the linear bushing's setpoint is also at a singularity (e.g. you attach the bushing between the world frame and a rotated frame of the body). I had encountered this in Anzu (issue 5446) as well.

EricCousineau-TRI, 2020-09-23:
Ah, to add to that - it's a bit painful b/c I need to now rotate the body+visual of the model itself so that the linear bushing itself is happy with the relative pose.

sherm1, 2020-09-23:
It should always be possible to provide an additional frame for the purpose of attaching a bushing. That frame should be oriented so that the pitch angle starts out near zero. You shouldn't have to rotate any physical objects to accommodate that -- it is really just a matter of how the bushing motion gets parameterized. Could you add another frame?

EricCousineau-TRI, 2020-09-23:
In my original usage in Anzu, I originally encountered this due to attaching that intermediate frame.

I've updated the repro, and you can see the diff here:
https://gist.github.com/EricCousineau-TRI/57cfaa3455d1e10255897a4253c0f2fc/revisions#diff-8db96eb2d46ecd776f16aa6b9a6661fd

This just sets `frameC` to incorporate the gimbal-locked pose; it still triggers the same error.

EricCousineau-TRI, 2020-09-23:
Ah, should I have offset frameA and frameC?

EricCousineau-TRI, 2020-09-23:
And nuts, my update was wrong. Lemme fix that right quick.

EricCousineau-TRI, 2020-09-23:
Yeah, this makes it work:
https://gist.github.com/EricCousineau-TRI/57cfaa3455d1e10255897a4253c0f2fc/revisions#diff-8db96eb2d46ecd776f16aa6b9a6661fd

Which is effectively what I had in Anzu.
I think my issue in Anzu was that I also had a contact-visualizer, which tickled this codepath when computing contact. At that point, I hadn't initialized `X_WO` to be at the setpoint, hence why I would get the singularity issue.

So yeah, it seems like if I just make sure to initialize that pose, I should be good.

Will add a recommendation to the top of this issue.

EricCousineau-TRI, 2020-09-23:
Hm... Sorry to throw a wrench in that, but I'm still observing that `X_WO` is actually sensitive to what frame `frameC` is set to (either offset from the body or not).

sherm1, 2020-09-23:
Not sure I'm following this yet. Looks like the bushing is between the World frame and a body frame. Better to set it up so that the bushing angles start out near zero by providing an offset frame for that purpose. One thing to check is to convert **back** from RotationMatrix to RollPitchYaw -- the mapping is not unique so you might not get back what you put in for extreme angles.

EricCousineau-TRI, 2020-09-23:
So digging into it a bit further, it looks like the gimbal lock error was actually caused by my simulation going unstable; occasionally, it would graze against other errors (`fn >= 0` assertions, or just spitting out NaNs). This error was sensitive to how I postured the object, which is why I was a bit confused about the behavior.

If I turn down the force/torque stiffness/damping, or increase the object's inertia, I can avoid those instabilities (of course).

Will see if I can reproduce it. Not sure what'll be actionable from there, though.

EricCousineau-TRI, 2020-09-23:
Here's the repro:
https://gist.github.com/EricCousineau-TRI/57cfaa3455d1e10255897a4253c0f2fc/a92019d9bc486f75eba62171203d1edeb55149b1
```
$ python3 ./drake_issue14114_repro.py 
X_WO: (array([ 0.,  0.,  0.]), array([  0.00000000e+00,   6.36110936e-15,   9.00000000e+01]))
RollPitchYaw::CalcRpyDtFromAngularVelocityInParent(): Pitch angle p = -89.9803 degrees is within 0.458371 degrees of gimbal-lock. There is a divide-by-zero error (singularity) at gimbal-lock.  Pitch angles near gimbal-lock cause numerical inaccuracies.  To avoid this orientation singularity, use a quaternion -- not RollPitchYaw. (bazel-out/k8-opt/bin/math/_virtual_includes/geometric_transform/drake/math/roll_pitch_yaw.h:323).
t: 0.0143
X_WO: (array([  3.77848269e-06,  -1.93509658e-06,  -9.63767921e-04]), array([-134.73665537,  -89.98030338,  -44.76860014]))
```

Yeah, this may just be chalked up to PEBKAC - should've checked inertias / stiffness, etc.
The error message was a bit confusing, but it's simulation instability, so dunno what can be clarified there.

Is there anything worth salvaging in this issue, or should it just be closed?

sherm1, 2020-09-24:
I would have liked the message to note that the error was reported by a bushing -- currently the low-level method that detected the error doesn't know who's asking. @mitiguy is back tomorrow maybe he can take a look to see if it would be feasible to get some more info squeezed into that message.

EricCousineau-TRI, 2020-09-24:
Per VC w/ @mitiguy, we briefly discussed that the error message could be implemented by checking for singularity in the linear bushing code itself, and then adding some additional semi-vague advice about checking your configuration as well as parameters affecting stability (stiffness, damping, mass / inertia, etc.).

mitiguy, 2020-09-26:
Opened PR #14126 to initiate progress on a better error message.

PR #14126 (2020-10-02): Versatile error message for gimbal-lock singularity (helps debug bushings).
Resolves #14114 which requested a better error message when users encounter gimbal lock during use of class LinearBushingRollPitchYaw.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14126)
closed 2020-10-02 by mitiguy