RobotLocomotion/drake #14118: __init__.py installed directly into /opt/drake/lib/pythonX.Y/site-packages

jamiesnape, 2020-09-24:
Does not seem right... Contents are `# Empty Python module `__init__`, required to make this a module.` which narrows the possibilities down to seven files, I think.

FYI @EricCousineau-TRI

EricCousineau-TRI, 2020-09-24:
Aye. Perhaps an easy debugging step is to teach `install.py.in` to annotate the files somehow.
e.g. "if it ends with `.py`, inject source location", or just "add `{file}.meta.txt` which indicates where it comes from".

jamiesnape, 2020-09-24:
I would put a fair amount of money on it being `lcmtypes/__init__.py`.

jwnimmer-tri, 2020-12-10:
Yes, I added a debug print and I see `-- Installing: lcmtypes/__init__.py to /home/jwnimmer/tmp/bung/lib/python3.6/site-packages/__init__.py`.

PR #14444 (2020-12-10): lcmtypes: Don't install stray __init__.py into site-packages
Closes #14118.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14444)
closed 2020-12-10 by EricCousineau-TRI