RobotLocomotion/drake #14121: Support an open source SOCP solver

hongkai-dai, 2020-09-25:
Currently I found open-source solvers 

- [ecos](https://github.com/embotech/ecos) This uses GPL license, hence not usable for us.
- [socp](https://web.stanford.edu/~boyd/old_software/socp/). This solver is >20 years old and hasn't been maintained/updated. @jwnimmer-tri and @jamiesnape , could you check if this solver has a license issue? 
- SCS. It is already supported in drake, but SCS doesn't always compute accurate solution.

jwnimmer-tri, 2020-09-25:
> This solver is >20 years old and hasn't been maintained/updated.

I was initially confused about the link, since it was only matlab code; but then later I found that tarball had a h/c file inside of it.  Are those the files you'd want to use?

> ..could you check if this solver has a license issue?

Its license looks like an ancient version the MIT license.  It's probably okay -- it forces the notice into the _documentation_ specifically, which is a bit more strict than normal.  Modern licenses talk about notice alongside the distribution, instead of literally in the documentation.

hongkai-dai, 2020-09-26:
> Are those the files you'd want to use?

Yes that tarball h/c file is what I want to use.

jamiesnape, 2020-09-28:
The license seems fine, though obviously the documentation clause is a little annoying. The MIT license is not really standardized, so there are lots of variations (though most people use the Expat variation). Looks like similar licenses made it into Debian, and they are usually pretty strict.

jwnimmer-tri, 2023-05-22:
For clarity, I think #19449 would be one solution to this request.  Is that correct?

hongkai-dai, 2023-05-23:
> For clarity, I think #19449 would be one solution to this request. Is that correct?

That is right.

hongkai-dai, 2024-06-25:
Close this as we support Clarabel.

closed 2024-06-25 by hongkai-dai