RobotLocomotion/drake #14138: Context::SetTime() and Context::SetAccuracy() should be allowed to be called on non-root context

xuchenhan-tri, 2020-09-28:


sherm1, 2020-09-28:
This was motivated by a @siyuanfeng-tri use case: he had a diagram ("old") and wanted to make a change to _one_ of its subsystems in an otherwise identical diagram ("new"), then transfer the state from old to new. He couldn't use SetTimeStateAndParametersFrom() at the root level because of the changed subsystem. But he could use it for each subsystem _except_ the changed one, which he could transfer by hand. That didn't work because the attempt to set Time and Accuracy was forbidden at a non-root level, even though he wasn't actually changing those.

jwnimmer-tri, 2020-09-28:
The `SetTimeStateAndParametersFrom` is just sugar (with epsilon-more-efficient ticket invalidation).  If the user wants to opt-out of some parts of the sugar, isn't this clearer anyway?
```
foo_context->get_mutable_state().SetFrom(bar_context.get_state());
foo_context->get_mutable_parameters().SetFrom(bar_context.get_parameters());
```
If we're not going to copy the time or accuracy anyway, maybe we shouldn't be trying to call a method that claims to do so?

I'm skeptical about adding more complexity to the contracts of SetTime, SetAccuracy, and SetTimeStateAndParametersFrom that carves out a special case for this.  The special case would have to speak to ticket invalidation ripples for "happened to be the same time", as well as whether two AutoDiff times are the same if the value the same principal value, or also if their derivatives must match.  The latter comparison is not what `operator==` get you.

sherm1, 2020-09-28:
> If we're not going to copy the time or accuracy anyway, maybe we shouldn't be trying to call a method that claims to do so?

The proposal would still meet the contract for the API -- the resulting time & accuracy would be "as if" the copy had occurred. Any attempt to actually make a change would fail if not done at the root level. Also the contracts for [SetTime()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_context.html#a3d5dddf9b2bb3283f9cd74c7a77517fc) and [SetAccuracy()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_context.html#a82e68f81f3284c08aa1c9acbfabbe1f5) promise to perform cache invalidations "at least if the setting has actually changed" so no further elaboration would be required. (We did that intentionally to leave open the possibility we would suppress these invalidations as a future optimization.)

This change just makes the sugar method more usable. It would have worked exactly as Siyuan expected it to with this change and he never would have had to think about what it was doing. As it is it was an impediment to him getting done what he needed to.

jwnimmer-tri, 2020-09-28:
If I call `Context<AutoDiffXd>::SetTimeStateAndParametersFrom` on a subcontext with a current time of `{.value = 10, .derivatives = [1]}` passing in a subcontext with a new time value of `{.value = 10, .derivatives = empty}`, what should happen?

sherm1, 2020-09-29:
That should fail. But I see your point -- `{10,[1]} == {10,[]}` would return true since it depends only on the value part of an AutoDiffXd. We would need an `is_exactly_the_same(t1, t2)` that can work for double, AutoDiffXd, and symbolic. The distributed time in a Context has to behave as though there was exactly one time entry shared by all subcontexts. Does such a method currently exist?

(This doesn't apply to accuracy which is always a double if it is present at all.)

jwnimmer-tri, 2020-09-29:
No, I don't think we have a generic "are two scalars structurally equal" checker yet.  We have [EqualTo](https://drake.mit.edu/doxygen_cxx/classdrake_1_1symbolic_1_1_expression.html#abb771e5bf5c37eb4753daf77f77adab0) for symbolic, and would have to write our own for `AutoDiffXd` (I think).

Note that `SetTimeStateAndParametersFrom` promises to discard all derivatives (including of the `get_time`) from the new incoming context, so we're either going to have a weird one-sided allowlist for when we treat almost-alike time values as sufficiently alike to allow subcontexts to be provided, or else we'll have to add an carve-out to the "discards derivatives" rule.  It doesn't actually _call_ `SetTime`, so we can allow them to have different API contracts if we wish.  There is already predecent for that with one demoting to `double` inbetween, and the other one not.

I'll grant that that fixing `SetTime` to short-circuit is not extraordinarily complex in the overall picture, but on the other hand `SetTimeStateAndParametersFrom` doesn't call it so that wouldn't actually fix the users' surprise.  Giving `SetTime` a short-circuit just makes a simple method more complex.  Focusing on a new contract for `SetTimeStateAndParametersFrom` is probably more worthwhile.  Trying to write up its new proposed Doxygen would be a worthwhile discussion.

sherm1, 2020-09-29:
> Note that SetTimeStateAndParametersFrom promises to discard all derivatives (including of the get_time) from the new incoming context ...

Where is that promise made? I don't see it in [doxygen](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_context.html#ae55845b172939052d02ea437df63cfbc). I see that [ValueConverter](https://drake.mit.edu/doxygen_cxx/structdrake_1_1systems_1_1scalar__conversion_1_1_value_converter.html) says it might discard derivatives, but it also says the scalar is unchanged if type U == T.

sherm1, 2020-09-29:
Are you interpreting "Sets this context's time, accuracy, state, and parameters from the double values in source ..." to mean _only_ the double values in source? I suspect when that was written we were assuming that the source would always be U==double.

jwnimmer-tri, 2020-09-29:
Oh, you're right.  I took that documentation at its word today, but it looks like I left the documentation a bit of a mess when I added `typename U` there (and elsewhere), so it was lying.  (The method _previously_ only accepted `Context<double>`.)

So it would be slightly less arcane than I'd been anticipating to adopt this short-circuit in the three methods in question.

Here's an alternative and simpler, proposal though:
- Add new `Context::SetStateAndParametersFrom` method.
- Change the docs and exception message from `SetTimeStateAndParametersFrom` to refer users to that method when dealing with subcontexts.

That has the additional advantage of not implying that time might be changing.

sherm1, 2020-09-29:
That does seem nicer!

xuchenhan-tri, 2020-09-29:
Thanks for the discussion and suggestions! I agree that adding `Context::SetStateAndParametersFrom` method is indeed cleaner. I will do that instead.

What is the standard procedure to reflect the result of this discussion? Should I modify this issue or close it and open a new one?

sherm1, 2020-09-29:
The best thing to do is edit the title and description if necessary but otherwise keep the issue. The discussion here is valuable and worth keeping around. The description for the PR for SetStateAndParametersFrom() should say it "Resolves #this_issue_number" which is GitHub magic that will cause this to close automatically when the PR merges. (You can also say "Fixes" or "Closes".) That links the PR to the discussion from which it arose which is good history.

cc @siyuanfeng-tri any thoughts on this?

siyuanfeng-tri, 2020-09-29:
I like @jeremyma-tri 's proposal. That's essentially what I am doing now. Thanks guys.

PR #14156 (2020-10-05): Add Context::SetStateAndParametersFrom()
Resolves #14138.

Add the method Context::SetStateAndParametersFrom() to allow
copying state and parameters from another context to a non-root
context.

Change the documentation for
Context::SetTimeStateAndParametersFrom() to reflect what it does.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14156)
commit f98ccdc2 (2020-10-05): Add Context::SetStateAndParametersFrom() (#14156)

* Resolves #14138.

Add the method Context::SetStateAndParametersFrom() to allow
copying state and parameters from another context to a non-root
context.

Change the documentation for
Context::SetTimeStateAndParametersFrom() to reflect what it does.
closed 2020-10-05 by SeanCurtis-TRI