RobotLocomotion/drake #14157: Accept patch-level variation for GUROBI_HOME automatically

jwnimmer-tri, 2020-10-02:
As of today, Drake is hardcoded to use _only_ Gurobi version 9.0.2 (when enabled), but not all users will be able to easily obtain that exact version match.  The download link from Gurobi's website is currently for 9.0.3.

Because Gurobi is commercially licensed software, users will typically bring their own copy (we can't provide one -- though see #10804 for related), so therefore it is difficult for us to control which version they are bringing.  The Gurobi website also tends to remove the download links for prior releases (especially patch releases), so it's not obvious how to obtain the exact revision that Drake wants.  (The trick is to just edit the download URL.)

We're already taking `GUROBI_HOME` as input from the user at fetch-time to know which Gurobi to use.  It would be nice if any matching patch-level release (9.0.x) could be provided without complaint.  As it stands now, we see:

```
Could NOT find Gurobi: Found unsuitable version "9.0.3", but required is exact version "9.0.2" (found /opt/gurobi903/linux64/include)
```

\CC @hongkai-dai

jwnimmer-tri, 2020-10-02:
This is conceptually easy, but a lot of boilerplate to wade through.  The gurobi library's filename, packaged install location, etc. all incorporate the version number and are referred to in dozens of place by our CMake and Bazel integration alike.

jwnimmer-tri, 2022-12-08:
Drake currently supports 9.5.1.  Per [here](https://support.gurobi.com/hc/en-us/articles/360048138771-Gurobi-release-and-support-history) we can see that 9.5.2 is already available.  The call to action here is a PR to Drake so that **either** 9.5.1 or 9.5.2 can be used for the same revision of Drake when building from source, without any code changes.

jwnimmer-tri, 2023-01-10:
+@BetsyMcPhail as a final item to prove that this patch works, please upgrade our CI images to use Gurobi 9.5.2 instead of Gurobi 9.5.1.

BetsyMcPhail, 2023-01-11:
For reference, #16604 updated Gurobi to 9.5.1. The checklist from that issue is below:

- [ ] ~_Update required version in Drake itself_~ should be covered by changes in this PR
- [ ]  ~_Update Gurobi cask_~ for drake_visualizer, no longer supported on mac
- [ ]  _Update Drake CI scripts_ 
- [ ]  _Update license server_ I am unsure if we need to regenerate the license for 9.5.2 vs. 9.5.1?
- [ ]  _Update CI provisioned Mac and Linux images_

BetsyMcPhail, 2023-01-21:
@svenevs and @williamjallen when you update the Linux AMIs for #18617 next week, do you want to also update the Gurobi version on the provisioned images?

williamjallen, 2023-01-21:
That seems reasonable to me.  I was going to shadow Stephen on Monday anyway, and needed to ask him about updating Gurobi at the same time so we can knock out two things at once.

BetsyMcPhail, 2023-01-23:
The last step here is to update the Gurobi cask and the Mac images. @svenevs and @williamjallen can you please work together on this.

svenevs, 2023-01-23:
Gurobi cask is updated and live at `robotlocomotion/director`.  The x86 monterey image has been updated to include gurobi 9.5.2.  Orka (arm64) update is planned for Wednesday.

BetsyMcPhail, 2023-01-26:
Orka has also been updated. Closing this issue.

closed 2023-01-26 by BetsyMcPhail