RobotLocomotion/drake #14197: Use convex hull approximation as default collision geometry for mesh files

RussTedrake, 2020-10-12:
I am following up on the discussion from [stackoverflow](https://stackoverflow.com/questions/64196799/mesh-based-collision-geometry-in-drake).  I believe that the fact that, by default, specifying a mesh file as a collision geometry results in NO collision geometry represents a *major* usability issue.  For newcomers, they immediately get a bad impression of our dynamics engine.  For experts, it’s a serious limitation to have to curate the collision geometry ourselves.

@sherm1 suggested that bullet is likely using the convex hull.  Browsing the bullet source code quickly, this seems likely, (and I’ve heard the same sentiment from a number of people) but I haven’t officially confirmed that.

Is there any reason why we shouldn’t compute the convex hull and use it as the default collision geometry (printing a warning to the log)?

avalenzu, 2020-10-12:
+1 from me. That's the behavior we used to have when we used Bullet.

SeanCurtis-TRI, 2020-10-12:
Agreed. W.r.t. bullet, we got the convex hull for free (buried inside the Bullet API). We don't have that anymore. Now we need to compute the hull ourselves in Drake.

Suggestions for libraries with compatible licenses? Qhull?

SeanCurtis-TRI, 2020-10-12:
Actually, if we're willing to take a performance hit, I believe there's a way in which we *could* move forward without increasing our external dependencies. I'll have to ponder it.

sherm1, 2020-10-12:
The Bullet license is compatible with Drake's so we could extract the convex hull code from there if that makes sense.

SeanCurtis-TRI, 2020-10-12:
It may not compute the convex hull directly. It may have an on-the-fly representation of the convex hull based on the set of vertices that are compatible with its GJK representation.

avalenzu, 2020-10-12:
@SeanCurtis-TRI is correct. We just fed the vertices of the mesh into their GJK API, which gave us signed distances to the convex hull of those points.

RussTedrake, 2020-11-13:
I'm going to upgrade this to high priority, as I've had about 5 more questions / comments about it from very different user groups since the issue was opening.  It is proving to be a major pain point for users.

PR #14351 (2020-11-20): [proximity] Implement Mesh for proximity via its convex hull.
This provides an initial implementation of the Mesh shape for proximity queries. It does so by using a very simple and very slow representation of the mesh's convex hull.

1. Refactor the OBJ --> fcl::Convex code to support Mesh. This basically *moved* the functions so they'd be available more generally.
2. Remove the all artifacts of special mesh treatment:
   - MeshIdentifier reifier.
   - special dynamic_mesh_tree_, anchored_mesh_tree_, and X_MeshBs_ (no longer necessary because the meshes don't have to be segregated).
3. Reimple [... cut]
closed 2020-11-20 by sherm1