RobotLocomotion/drake #14212: common: Should define clear convention on how to handle points

EricCousineau-TRI, 2020-10-14:
Namely, should directories always end with slashes, or never. It should never be "in between".

Example: `PackageMap::GetPath()` should either always have a trailing slash, or never. Period.
It should either fail if `Add()` is called poorly, or it should strip it.

See:
https://github.com/robotlocomotion/drake/pull/14211#pullrequestreview-508375268

jwnimmer-tri, 2021-11-12:
I don't necessarily disagree, but:

(1) Do we need a tracking issue for this, given that nobody is ever going to work on it?

(2) Once [std::filesystem](https://en.cppreference.com/w/cpp/filesystem) support is added by Apple Clang, we should just change all of our std::string paths to fs::path paths, and then the problem is resolved.

Given that, should we close this issue?

EricCousineau-TRI, 2021-11-12:
Yup, SGTM!

closed 2021-11-12 by EricCousineau-TRI