RobotLocomotion/drake #14215: autodiff speedups via pool allocator

rpoyner-tri, 2020-10-15:
As of this [writing](https://github.com/RobotLocomotion/drake/issues/13902#issuecomment-707332636), AutoDiffXd operations on master are still thrashing the heap too much. This issue is to track solution(s) to that specific problem.

Two approaches are currently under investigation:
 * Use a VectorXd-compatible type with internal storage allocated via pool (Sherm's pool_vector_autodiff_hacks branch)
 * Use an Eigen vector type allocated-entire via pool (Rico's vector-pool branch, TBD)

One or both of these approaches should be able to work; Sherm's approach has the potential to allow further instruction-level optimizations, since it takes local control over more of the vector math.

rpoyner-tri, 2020-10-22:
Here's some performance data for my experimental pool PR at #14242.

Comparing master (341b6a7f6) to `r1` of the PR (https://github.com/rpoyner-tri/drake/commit/0e4e00e63baf7b8263a7c01adda02a0ebe5f2085), on my Puget:
[image]

So, we can see `r1` gives ~15-20% speed-up compared to contemporary master. See the commit log message for some of the trade-offs taken to get here.

rpoyner-tri, 2020-10-23:
At @jwnimmer-tri 's excellent suggestion, I measured the memory use of the experimental branch vs. master for `cassie_bench`. Oh the hilarity.

Here's a plot of memory use from master:
[image]

And here's from `r1` of the PR at #14242:
[image]

That's something like a 15x increase in memory use! Taking the advice to avoid allocating for scalars without derivatives will probably help; only a proper experiment will tell.

The above plots were generated using `apt install python3-memory-profiler`. Just do `python3-mprof run [PROGRAM]; python3-mprof plot`. The result of `plot` is an interactive plot window; from there you can save a png file. The tracked measurements are resident set (RSS) only.

rpoyner-tri, 2020-11-15:
Starting to roll a PR train, based on the full dev branch available at #14318. The speedup is in the ballpark of 10-15% better than current master on cassie_bench AutoDiff cases. The memory usage is about 15% worse than master on the benchmarks.

There is a further tweak to this branch that could squeeze out another maybe 5% performance improvement, but the style guide forbids it. This would be to use the `__thread` GNU extension for thread-local storage in a few places. It has the effect of avoiding lazy-init wrappers provided by the compiler for c++11 `thread_local` in places where the current design does not need or want them. In local testing, it compiles and runs properly on all currently supported compilers.

jwnimmer-tri, 2020-11-16:
Before the PR train invests in code review and polishing -- the first question to speak to for all of your reviewers should be whether a 15% speedup is worth the complexity cost of the implementation, both in terms of ongoing maintenance and upkeep, and also in terms of everyday users being able to debug heap-related problems.  When I was prototyping the dense storage override to try SBO out for small AutoDiff programs, I was hoping for 2x-4x speedup, not 1/6th speedup.

rpoyner-tri, 2020-11-16:
Of course it's not worth it. But I need to offer a plausible attempt in order to falsify the hypothesis, lingering here for months, that removing heap thrash would clear the way to substantial speedups.

It's possible that some sequence of optimizations, including this one, will yield a roughly 2x speedup of cassie_bench from the original baseline. Beyond that, i think we hit the wall. If we really want fast differentiation, we would have to invest much more in some completely different strategy. I just don't believe that forward autodiff as implemented in drake (deliberately disabling eigen expression tree optimizations) will ever get there.

sherm1, 2020-11-16:
I'm confused about the purpose of the PR train then. The return value/move optimizations you put in master seemed well worth it. But if this much more complicated revision is only a 15% further speedup that doesn't seem worth the review effort since I presume we wouldn't want that in master. Can you clarify, or should we discuss f2f?

rpoyner-tri, 2020-11-16:
I thought it was obvious to all that this path wasn't going to get another 2x. Even abandoning all thread safety would only get you to about 25% better at the outside. I guess the part I'm confused about is where your cutoff is. If the next change has to achieve 2x beyond current master, we should just stop now - no such improvement is forthcoming in the current forward autodiff scheme.

sherm1, 2020-11-16:
Per f2f -- more of a cost/benefit determination, and a fuzzy one at that. A small change that gave 15% speedup would be a clear win, but that same speedup from a substantial change like this one is likely not a good tradeoff. Also an architectural change that didn't pay off immediately but laid groundwork for substantial speedups in follow-on work could be worth doing (doesn't apply here as I understand it).

rpoyner-tri, 2020-11-16:
I've posted a few final tweaks to the full-branch PR. At this point I'm ready to tap out and move on to something else.

Some final numbers:
[image]

closed 2020-11-16 by rpoyner-tri

sherm1, 2020-11-16:
Are those final numbers just 15% better than master's?

rpoyner-tri, 2020-11-16:
Let's say 18-20%, but yes, that is compared to fairly recent master.