RobotLocomotion/drake #14224: rpath set correctly only upon "make" and not "make install" from CMake build

edrumwri, 2020-10-18:
If I run `cmake -DCMAKE_BUILD_TYPE=Debug $DRAKE_DIR` and then build with `make`, my libraries' rpaths will be set correctly: I run `lcm-gen` from my local install binary directory and it works properly. If I instead build with `make install`, my libraries' rpaths are not set correctly: running `lcm-gen` from my local install binary directory results in:

    lcm-gen: error while loading shared libraries: libdrake_lcm.so: cannot open shared object file: No such file or directory

This seems to only happen in the Debug build (?!), though don't quote me on that.

I've traced the offending action down to the following command being run during `make install`:

    make -f CMakeFiles/drake_cxx_python.dir/build.make CMakeFiles/drake_cxx_python.dir/build

jwnimmer-tri, 2020-10-19:
Can you provide instructions on how to reproduce the bug?

It seems to work fine for me:

```console
jwnimmer@call-cps:~/tmp$ cat repro.sh 
#!/bin/sh
set -ex
cd /home/jwnimmer/tmp
rm -rf build install
mkdir -p build build/drake-build install
cd build
wget https://github.com/RobotLocomotion/drake/archive/v0.23.0.tar.gz
tar xfz v0.23.0.tar.gz
cd drake-build
cmake ../drake-0.23.0 -DCMAKE_INSTALL_PREFIX=/home/jwnimmer/tmp/install
make install
cd /home/jwnimmer/tmp
/home/jwnimmer/tmp/install/bin/lcm-gen --help

jwnimmer@call-cps:~/tmp$ ./repro.sh 
+ cd /home/jwnimmer/tmp
...
[100%] Completed 'drake_cxx_python'
[100%] Built target drake_cxx_python
Install the project...
-- Install configuration: "Release"
-- Installing: /home/jwnimmer/tmp/install/share/doc/drake/VERSION.TXT
+ cd /home/jwnimmer/tmp
+ /home/jwnimmer/tmp/install/bin/lcm-gen --help
Usage: /home/jwnimmer/tmp/install/bin/lcm-gen [options] <input files>
...

jwnimmer@call-cps:~/tmp$ readelf -d install/bin/lcm-gen

Dynamic section at offset 0x36a58 contains 32 entries:
  Tag        Type                         Name/Value
 0x0000000000000003 (PLTGOT)             0x36d00
...
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libdrake_lcm.so]
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
...
 0x000000000000001d (RUNPATH)            Library runpath: [$ORIGIN/../lib]
...
```

jwnimmer-tri, 2020-10-19:
I also tried the same using Debug build in the repro script (i.e., `cmake ../drake-0.23.0 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/home/jwnimmer/tmp/install`), with the same result.

edrumwri, 2020-10-19:
Looks like `make` before `make install` is what does it.

I did this:

```
drum @ caesar /tmp
└─ $ ▶ cat repro.sh 
#!/bin/sh
git clone https://github.com/RobotLocomotion/drake
cd drake
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Debug ..
make
make install
./install/bin/lcm-gen
```

And got this from readelf:
```
drum @ caesar /tmp
└─ $ ▶ readelf -d ./drake/build/install/bin/lcm-gen 

Dynamic section at offset 0x45a40 contains 32 entries:
  Tag        Type                         Name/Value
 0x0000000000000003 (PLTGOT)             0x46ce8
 0x0000000000000002 (PLTRELSZ)           2304 (bytes)
 0x0000000000000017 (JMPREL)             0x1730
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000007 (RELA)               0x1418
 0x0000000000000008 (RELASZ)             792 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffff9 (RELACOUNT)          22
 0x0000000000000015 (DEBUG)              0x0
 0x0000000000000006 (SYMTAB)             0x298
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000005 (STRTAB)             0xce8
 0x000000000000000a (STRSZ)              1493 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x12c0
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libdrake_lcm.so]
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000c (INIT)               0x2030
 0x000000000000000d (FINI)               0x372a0
 0x000000000000001a (FINI_ARRAY)         0x46a30
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x0000000000000019 (INIT_ARRAY)         0x46a38
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001d (RUNPATH)            Library runpath: []
 0x000000000000001e (FLAGS)              BIND_NOW
 0x000000006ffffffb (FLAGS_1)            Flags: NOW
 0x000000006ffffff0 (VERSYM)             0x12f8
 0x000000006ffffffe (VERNEED)            0x13d4
 0x000000006fffffff (VERNEEDNUM)         1
 0x0000000000000000 (NULL)               0x0
```

jwnimmer-tri, 2020-10-19:
Note that "make" actually does "make install" -- there is no difference.  I can confirm that this repro script shows the problem:

```bash
set -ex
cd /home/jwnimmer/tmp
rm -rf build install
mkdir -p build build/drake-build install
cd build
wget https://github.com/RobotLocomotion/drake/archive/v0.23.0.tar.gz
tar xfz v0.23.0.tar.gz
cd drake-build
cmake ../drake-0.23.0 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/home/jwnimmer/tmp/install
make install
make install
cd /home/jwnimmer/tmp
/home/jwnimmer/tmp/install/bin/lcm-gen --help
```

The bug is that Drake's `install.py.in` logic for rewriting rpaths fails to detect that `libdrake_lcm.so` is a candidate for rpath fixing, if we are _re-installing_ atop a prior install, and if we have _no changes_.  In release mode, there are always changes, so we avoid hitting the bug.  In Debug builds, the "don't re-copy files we already have" filecmp heuristic kicks in, and we lose the library from the `libraries_to_fix_rpath` set, so don't update `lcm-gen`'s rpath.

The work-around for now is to invoke make only once.  Or else, when calling it repeatedly, `rm -rf` the install tree prior to re-running make.

I'll ask Kitware to fix the logic error, but I'm not sure what that will happen.

jwnimmer-tri, 2020-10-19:
Specifically, the issue is the early return on L168, so that the assignment on L183 never happens, so the repair on L298 never happens.

https://github.com/RobotLocomotion/drake/blob/7b7bf9e95a88f4b03aca4905f1335625228a60ce/tools/install/install.py.in#L158-L183

https://github.com/RobotLocomotion/drake/blob/7b7bf9e95a88f4b03aca4905f1335625228a60ce/tools/install/install.py.in#L292-L298

jwnimmer-tri, 2020-10-19:
To @jamiesnape for delegation.  This shouldn't preempt your current board, but if there is anyone else at Kitware who has time to fix this and add an automatic regression test, that'd be helpful.

jwnimmer-tri, 2020-10-19:
Related to #7331.

jamiesnape, 2020-10-20:
#7331 is definitely overdue. My fantasy is that `cmake -P cmake_install.cmake` would work and/or CMake could (optionally) do the heavy lifting, but that probably will remain a dream.

mwoehlke-kitware, 2022-08-25:
> the issue is the early return on L168

This analysis (and the associated TODO) is AFAICT incorrect. We check if the presently installed artifact matches the as-built artifact, which means in practice a library (that has had non-no-op patching performed) will always be reinstalled.

I would expect multiple `bazel run //:install` to also show the problem if it's in `installer.py`. So far, no dice, but that was with no `-c`. Currently trying again with `-c dbg` (which will take some time). Not sure why this would *only* affect debug, but all the repro cases seem to be using debug, so...

mwoehlke-kitware, 2022-08-25:
Okay, I can reproduce by running the following twice:
```
bazel run --compilation_mode=dbg --fission=no //:install -- --no_strip ~/drake-install
```
This works *with non-CMake builds*. It *stops* working if I remove `--no_strip`.

mwoehlke-kitware, 2022-08-25:
Fix at #17779. (@jwnimmer-tri, feel free to take a look or even 'steal' the review if you like; I mainly tagged Rico because you're out.)

PR #17779 (2022-08-29): Fix install RPATH munging
Modify `installer.py` to separately track all installed libraries versus libraries which need RPATH fixing. The former includes libraries that were already installed ("up-to-date"), which still need to be considered when fixing RPATHs for anything being (re)installed.

This fixes RPATH breakage in some instances when installing to the same location multiple times, especially if nothing else has changed in the interim.

Fixes #14224.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17779)
commit 7a6921ee (2022-08-29): Fix install RPATH munging (#17779)

* Fix install RPATH munging

Modify installer.py to separately track all installed libraries versus
libraries which need RPATH fixing. The former includes libraries that
were already installed ("up-to-date"), which still need to be considered
when fixing RPATHs for anything being (re)installed.

This fixes RPATH breakage in some instances when installing to the same
location multiple times, especially if nothing else has changed in the
interim.

Fixes #14224.
closed 2022-08-29 by rpoyner-tri